OpenC3/cosmos #246: Cannot login to minio console using standard credentials (openc3minio, openc3miniopassword)

jurgensymynck, 2022-11-13:
(This problem is likely related to issue https://github.com/OpenC3/cosmos/issues/245)

See screenshot. I cannot login to minio console using standard credentials (openc3minio, openc3miniopassword)
I noticed that  the openc3-minio container does not have latest minio release ...

in the container:
```
sh-4.4# minio  
VERSION:
  RELEASE.2022-10-02T19-29-29Z
```

according to minio console:
```
Latest Version:
 minio/minio:RELEASE.2022-11-11T03-44-20Z
```

**Environment**
- windows 10
- OpenC3 Version 5.0.11
- Browser: Google Chrome
- firewall: windows defender, my wifi network is set to "private profile"
- docker: v4.14.0

[image]

ryanmelt, 2022-11-13:
In your screenshot, you are adding in a dash character to the username.
openc3-minio should be openc3minio

Also, we will be updating minio to the latest in our 5.1.0 release this week.

jurgensymynck, 2022-11-13:
Ah, I fell victim to Chromes Auto Filling of forms...
Will close the issue for now, verifying will have to wait until issue https://github.com/OpenC3/cosmos/issues/245 is resolved.

Thanks!
Jurgen

closed 2022-11-13 by jurgensymynck