OpenC3/cosmos #248: StreamingApi

mteckes, 2022-11-15:
I am attempting to determine when a test suite is completed without having a person watching / checking the test suite. I  was wondering if the StreamingAPI was the right way to accomplish this task. I have currently subscribed to the StreamingChannel, but have not seen any data come through besides the confirmation. Am I supposed to see anything else on that channel?

Thanks for the help!

ryanmelt, 2022-11-15:
Not the StreamingApi.  You would need to connect to the websocket channel used by ScriptRunner (RunningScriptChannel)

mteckes, 2022-11-15:
Thank you so much Ryan!!! 

I didn't realize there was another cable endpoint `/script-api/cable` 

If anyone else has questions on there, I was able to construct from [this](https://github.com/OpenC3/cosmos/blob/51d16bb293d5a9a762a5e0c00ebcfcfadb567120/openc3-cosmos-init/plugins/packages/openc3-cosmos-tool-scriptrunner/src/tools/ScriptRunner/ScriptRunner.vue)

closed 2022-11-15 by mteckes