OpenC3/cosmos #249: Connection Failed: ENOENT: No such file or directory @ rb_sysopen - COM21

Theskyfalls, 2022-11-15:
[image]

[image]

Hey I am trying to understand where to start troubleshooting with this error.  

Any help pointing in directions would be great.  

Thanks

ryanmelt, 2022-11-15:
Hello!

Serial ports don't work in Docker unless you expose them to the container (linux only).   

See: https://github.com/OpenC3/openc3.com/issues/16

Also - You should probably update to the latest OpenC3 :)

Theskyfalls, 2022-11-17:
Thanks For the info!   

So to confirm there is not a way to use serial ports on Windows through docker? 

Currently I am able to run cosmos on my Windows machine,  and unable to run it on the Linux VM I have set up

ryanmelt, 2022-11-17:
Sorry, for Windows you can use our bridge concept.  Basically its a small process that reads the serial port and forwards the data over ethernet to COSMOS.   

See: https://github.com/OpenC3/openc3.com/issues/16

closed 2022-11-28 by jmthomas

Theskyfalls, 2022-11-30:
I am trying to follow the instructions on the https://github.com/OpenC3/openc3.com/issues/16 you have linked for me.

I am running into problems trying to understand the instruction steps for steps 2 and 3.  

Where do I change these variables?  Is this within Cosmos once I run the http://localhost:2900 

set host environment variable RUBYLIB=<Path to cosmos/openc3/lib folder>
set host environment variable RUBYGEMS_URL=https://rubygems.org/

My goal is to send and receive packets that go through our COM21 USB port and then through a RS 422 converter. 
Thanks for your help!