OpenC3/cosmos #253: Load tool configuration file with script (automatically on start-up)

KodiakLiDAR, 2022-11-15:
Hello OpenC3 team,

I would like to load pre-saved configuration files on OpenC3 start-up. Essentially, I'd like to be able to load saved configuration files, such as **Telemetry Grapher** configurations or **Telemetry Viewer-Telemetry Screen** configurations, using a script in _any language_.

This would allow a .sh/.bat script to launch OpenC3 and load all system relevant tool configurations on start up.

Thank you very much for any and all guidance.

Kindly,
Kodiak

jmthomas, 2022-11-16:
Requires ticket #187 to enable local mode to store those configurations if you want to start up with that stuff loaded.

ryanmelt, 2023-02-16:
This will be implemented with enhanced local mode #187. Closing this ticket.

closed 2023-02-16 by ryanmelt