OpenC3/cosmos #259: Cannot build fresh openc3 init container

GTI-Dakota, 2022-11-16:
Hello, while attempting to build a new init container for a server, I am receiving the following error:

`Step 16/37 : COPY ./plugins/docker-package-build.sh ./plugins/docker-package-install.sh ./plugins/babel.config.js ./plugins/.eslintrc.js ./plugins/.nycrc ./`

`COPY failed: file not found in build context or excluded by .dockerignore: stat plugins/docker-package-build.sh: file does not exist
ERROR: Service 'openc3-init' failed to build : Build failed`

Is this related to the base container that is being pulled by the openc3 init container? My environment variables specify the tag as 5.0.11

Thanks,

Dakota

ryanmelt, 2022-11-16:
I think that is saying that those files are missing locally (ie. ./plugins/docker-package-build.sh is not present in your local folder).

GTI-Dakota, 2022-11-16:
I see. Is there a place I can download those required files? I do not see them on my other server I'm using as a reference. I see a /plugins/DEFAULT/ directory but there is no docker-package-build file in there

ryanmelt, 2022-11-16:
Should be in openc3-init/plugins (or openc3-cosmos-init plugins with our next release)
https://github.com/OpenC3/cosmos/tree/main/openc3-cosmos-init/plugins

GTI-Dakota, 2022-11-16:
I'll give it a try, thanks!

GTI-Dakota, 2022-11-16:
New error!

```
COPY failed: no source files were specified
ERROR: Service 'openc3-init' failed to build : Build failed
```

This is after downloading the plugins directory. For edification, I am using the enterprise init build. Not sure if this make a difference

ryanmelt, 2022-11-16:
You shouldn't have to download anything...  
What command are you running?

GTI-Dakota, 2022-11-16:
`sudo ./openc3.sh build`. It is possible that this repo is outdated and incompatible w/ the current build directives. I had copied the repo I'm currently using from my reference server, which is using the 5.0.11 build from last month. I think I will re-start from scratch and see how that goes 👍

closed 2022-11-18 by GTI-Dakota