OpenC3/cosmos #260: Json Channel

MTI-twalker, 2022-11-16:
The redis stream is extremely useful but am unable to access the created Topic using the current channels available from the front end (without a work around). 

Is it possible to have a JsonChannel added to the list that takes an argument of "topics" type list.

Maybe a new model of: 
RealtimeJsonThread < StreamingThread
 - That uses a thread_body of a new function redis_json_body which is not expecting a packet (handle_message) but just transmits the msg_hash.

I was able to create a work around by ensuring my redis stream hash's match the telemetry topics like:
```
json_data = {"whatever":"whatever"}
msg_hash = {
  :time => 123456,
  :packet_time => 123456,
  :stored => false,
  :target_name => "name",
  :packet_name => "subname",
  :received_count => 1,
  :json_data => JSON.generate(json_data.as_json(:allow_nan => true)),
}     
```
So the topic name was required to be in the format of '```#{scope}__DECOM__{name}__subname```'    
On the front end, adding a StreamingChannel subscription to TLM__name__subname with mode of "DECOM". 
Just would be nice to not have to have that work around. 

Maybe there is a better way to do this?

ryanmelt, 2022-11-16:
Could you explain what you are trying to do?
What topic do you want to expose?
All of the internal packet topics are already exposed through the StreamingApi so I'm confused.

MTI-twalker, 2022-11-17:
My use case is currently looking for a list of files on a specific minio bucket/folder (not the config bucket). I have a background task that is using list_objects_v2 and collecting that key list. I am then posting that list to a topic so that my front end can capture it. For this specific use case, it would be better to have an API (similar to script-api/scripts, except a specific folder and not *.rb) that can grab a specific bucket/list so that a front end tool could capture it. The goal would be to download that file through the tool instead of the needing to log into the minio browser. 
While I still would prefer a new API for this specific use case, I am still interested in being able to take an output (any json data) from a back end script even if it isn't a packet and utilize the websocket for communications with the front end.

ryanmelt, 2023-02-13:
We now expose a way to directory list all buckets in the API.  See Bucket Explorer.  Closing.

closed 2023-02-13 by ryanmelt