OpenC3/cosmos #270: HttpClientInterface and HttpServerInterface

ryanmelt, 2022-11-18:
Add an HttpClientInterface to support connecting to HTTP servers (like web pages, and other HTTP based apis)
Add an HttpServerInterface to support receiving Webhooks

bradenvirdellgenevatech, 2022-11-30:
I have created an http interface in the past currently it only handles get and post requests as that is all we needed it for. I believe it could be expanded upon relatively easily. If you are interested in looking at let me know I have received permission to share it.

ryanmelt, 2022-11-30:
Sure, you can submit a pull request or email it to us.   I'll at least use it for inspiration.  I want to do a trade between the various http clients and make sure we are still using the best one.

HoboKennyG, 2023-06-28:
> I have created an http interface in the past currently it only handles get and post requests as that is all we needed it for. I believe it could be expanded upon relatively easily. If you are interested in looking at let me know I have received permission to share it.

Any chance you'd be willing to share with me the interface you wrote?  I'm also looking to use Cosmos to connect to a device using http get and post requests.

PR #805 (2023-09-17): Http Interfaces
HTTP Interfaces!

closes #270
closed 2023-09-17 by ryanmelt