OpenC3/cosmos #273: Streaming API connection resets  (V5.0.11)

MTI-twalker, 2022-11-18:
During some testing of requesting the previous day's worth of packets for a specific packet (1 every 2 seconds so roughly 43200 packets), it was found that not all packets are received by the subscriber. It was also found that the amount of packets received was inconsistent. 

It appears that the cause of this is that the connection drops during the downlinking. (python output during testing)
```
Establish connection...
Run connection loop.
Connection established.
Welcome message received.
On receive callback set.
Create subscription on server...
type: (confirm_subscription)
Subscription confirmed.
... <- Printed total count of packets received every 3 seconds
33090
Run connection loop. <-this is where it interrupts
Connection established.
Welcome message received.
33090 Received
```

The javascript library console printed "connected" during a transmission which immediately hung instead of getting the "[]" at the end during a previous run that received all packets implying the same issue. 

This was tested with both python and javascript:
- https://github.com/omniploy/python-actioncable-zwei
- https://www.npmjs.com/package/actioncable "^5.2.6"

I did a few tests and got inconsistent logs:
I watched the cmd_tlm_api logs and got this: 
 - The container reports "1 objects remain in stream" followed by "Finishing stream for topic:... - End of files"}". This was prior to all packets being received. (It continues with "Finishing 1 objects from stream", "0 objects remain in stream", "Sending stream complete marker")
 - The container reports "Finishing 1 objects from stream" followed by "0 objects remain in stream". This was prior to all packets being received. (It continues with "Sending stream complete marker")
 - The container reports "Finishing 1 objects from stream" followed by "Started GET "/openc3-api/cable"". This was prior to all packets being received.

Each request was identical. The 3rd being the most clear of a reset of connection. 1st and 2nd, the connection reset but I believe the stream finished prior to the new request coming in since almost all the packets were received (stopped at 40682). Unsure why the outcomes are different though.

So it appears the server is not aware of the dropping connection. The python configuration is: 
```self.websocket.run_forever(ping_interval=5, ping_timeout=3, origin=self.origin)```
This uses the websocket python library.
I was able to generate a connection and create a subscription without adding to the subscription and let it sit. The connection never dropped/reset.

ryanmelt, 2022-11-22:
Please try with 5.1.1.    The streaming api was completely refactored and should be much more reliable.

ryanmelt, 2022-11-27:
Raw issue is definitely a bug we'll get fixed.
Are your remaining ping issues just from Python?

MTI-twalker, 2022-11-28:
The ping issues are with any connection that expects to test the connected status while receiving data. I believe the client is expecting a ping from the server to make sure its still connected but isn't. I believe it is related to this:

https://github.com/rails/rails/issues/42336

Currently your configuration sets the stale connection to 60 seconds which is enough time for my requests but if it were longer than 60 seconds the client would reconnect most likely. Change that to 5 seconds and I would expect you see the same issues.

So what I did as a work around is change that to 60 in my configuration (libraries default was 6) and remove the ping expectation from the python configuration and I get the data.

ryanmelt, 2022-12-12:
Updated ping logic in #324

closed 2022-12-12 by ryanmelt