OpenC3/cosmos #277: Issue uploading Gems on 5.1.0 and 5.1.1

GTI-Dakota, 2022-11-21:
Uploading a Gem on COSMOS 5.1.0. and 5.1.1. causes a 500 error with an unknown reason.

[image]

I have built the gem with the newest download of the cosmos-project and also have tried uploading gems built on 5.0.11 with no luck. Is there a place I can check the logs to diagnose this issue? Or is there a new keycloak setting I am not aware of that needs to be updated?

Thanks,

Dakota

ryanmelt, 2022-11-21:
Hey Dakota - 
Can you exec into the cmd-tlm-api container and see who what your /gems/cache folder is owned by?

```
docker exec --user root -it <container hash here> sh
cd /gems
ls -la
```

everything should be owned by openc3/openc3

if not use chown to fix.

chown openc3:openc3 *

ryanmelt, 2022-11-22:
Oh - I think you are using a modified openc3-base container?

If so, you MUST make sure that the Dockerfile for the container finishes up running as a regular user and not root

USER ${USER_ID}:${GROUP_ID}

closed 2022-11-22 by ryanmelt