OpenC3/cosmos #287: COBS and SLIP protocol support for secure framed serial communication

jurgensymynck, 2022-11-27:
**Problem description**

I believe OpenC3/COSMOS does not support COBS or SLIP encoding/decoding, for reliable transfer of a frame of arbitrary bytes over serial connection. This is essential for reliable data transfer, especially at higher data rates and frame packet sizes.
As long as data rates are modest (25 Hz), packet size are small (30 bytes) and thus having a low chance of having a PACKET_ID byte inside of the frame payload, things work reasonably well without much data loss (I use LENGTH as packet delimiting for now)

After much reading (see Further reading) , I concluded that reliable serial data transfer proves problematic for a great number of embedded hardware users (including myself), and implementing this feature in COSMOS, together with using established libraries like [Serialtransfer](https://github.com/PowerBroker2/SerialTransfer) or [PacketSerial](https://github.com/bakercp/PacketSerial ) would open up COSMOS to a whole new userbase of advanced Embedded enthusiasts, and establish COSMOS as the go-to solution for hassle free communication with these embedded systems.

**Describe the solution you'd like**
- Create helper protocol for encoding/decoding COBS encoded serial data 
- Create helper protocol for encoding/decoding SLIP encoded serial data 

If this is not feasable, provide some guidelines to use existing Ruby gems in creating these helper protocols.

**Further reading**

- [Rubygerms.org](https://rubygems.org/search?query=serial) already has gems for parsing Serial data ([Packet-protocols](https://rubygems.org/gems/packet-protocols), [P3]( https://rubygems.org/gems/p3)), 
- Serial communication tutorial with problem descritipion: Serial input [Basics ](https://forum.arduino.cc/t/serial-input-basics-updated/382007 )and [Advanced](https://forum.arduino.cc/t/serial-input-advanced/634430)

>

ryanmelt, 2022-11-27:
We'll add this to our list.  Protocols are easy to create.  See https://openc3.com/docs/v5/protocols.

Serial connections definitely requires some form of framing.  I generally recommend just adding a sync pattern to the beginning of your packets which COSMOS supports out of the box.

ryanmelt, 2023-04-23:
I'm going to do the COBS protocol too.

closed 2023-04-23 by ryanmelt

jurgensymynck, 2023-04-24:
Thanks a lot! Looking forward to it.

PR #636 (2023-04-23): COBS Protocol
closes #287