OpenC3/cosmos #288: TCP Keep Alive Option

ryanmelt, 2022-11-28:
Something like: https://github.com/BallAerospace/COSMOS/pull/1800 except make an option

MTI-twalker, 2023-02-10:
I am not sure how you plan to implement the tcp keep alive ping. 

Possible additional enhancement:
The TCP connections currently go stale after a unknown amount of time. Days later, when attempting to use the connection the interfaces/routers state "CONNECTED". The way I can tell this is an issue is the connect/disconnect button no longer does anything. So the plugin needs to be reloaded. It would be good to auto_disconnect prior to the TCP timeout. That way at least you can reconnect without reloading.

ryanmelt, 2024-01-07:
TCP connections going stale is a known Docker Desktop issue that is discussed on our Installation page on how to address: https://docs.openc3.com/docs/getting-started/installation

PR #1018 (2024-01-08): Add TCPIP keepalive options
closes #288
closed 2024-01-08 by ryanmelt