OpenC3/cosmos #290: Best practice for separating openC3 from project specific git repository?

WillMarchant, 2022-11-28:
We are not yet ready to freeze on a specific release of openC3 and so want to be able to pull updates.

But we would like to use git to keep track of our project specific setup files.  In the example documentation you end up with openc3-cosmos-bob/ and plugins/DEFAULT/openc3-cosmos-bob/ as untracked files.

What are people doing about getting those untracked files checked into git?  Are you checking them into a separate git repository?

Thanks in advance for any advice!

ryanmelt, 2022-11-28:
You should clone this project: https://github.com/OpenC3/cosmos-project
Run from there and checkin everything from there to your own git repo.   (including everything in the plugins folder)

ryanmelt, 2022-11-28:
Note this will capture your main compose.yaml configuration, environment variables, and plugins installed.   It is currently missing user specific saved configurations for tools.  We'll be adding support for saving off those configurations as well soon.