OpenC3/cosmos #301: Sending a command (in a being rescue block) to a disconnected interface now crashes the interface

GTI-Dakota, 2022-12-01:
Hello,

Since the 5.1.0 update, gems that previously utilized microservices to poll telemetry within a begin rescue loop can no longer do so safely when an interface is disconnected. In 5.0.11, the rescue would handle a disconnected interface, but with the current build, the rescue is not handling it, and the interface crashes and must be spun up again. Thus, this creates an endless crash-restore loop as long as a particular device is not actively connected to.

[image]

We had stopped using the get_interface API call previously to check connected/disconnected state, but stopped using it due to server optimization reasons. Is it feasible to re-implement this API call once more in multiple microservices?

ryanmelt, 2022-12-01:
I can't recreate this.   

```
begin
  cmd("EXAMPLE START")
rescue
  puts "Rescued"
end
```

Works as expected.   The interface prints an error but does not crash.

Is this something with a custom interface you have built?

GTI-Dakota, 2022-12-01:
Oh, you're correct. My mistake, I had been looking at two errors, the one which was crashing the interface is not related to this issue and has been since resolved. I do see the large error in the log window.

Is there any way to suppress this error? It is quite large in the log window screen. I would expect the rescue to prevent such an error from printing, as it did in 5.0.11

ryanmelt, 2022-12-02:
Doesn't really need to post a stack trace for an interface not connected error.  I'll clean this up in the next release.

closed 2022-12-02 by ryanmelt