OpenC3/cosmos #302: Docker Unable to Start

MTI-GMays, 2022-12-01:
Docker will not restart after the containers are removed. After the './openc3 stop' command removes the containers and the run or start commands are used, the containers do not stay started. The operator, script runner, and the cmd-tlm-api container will continually restart. A clean up and prune were done to no avail. 

Environment
- OS: Windows 10
- Version: Cosmos 5.1.1
- Browser: Google Chrome

[image]

ryanmelt, 2022-12-01:
The init container has to run successfully. If it is restarting other things tend to have a bad day.

First step to debug is to look at the init container logs.

```
docker logs cosmos-511-openc3-cosmos-init-1
```

Let us know what that says.

ryanmelt, 2022-12-03:
If you upgraded from an earlier version, you may be running into an issue with the demo plugin being renamed in the latest version.    

If that is the issue, you can:
1. Remove the OPENC3_DEMO line from your .env file
2. Restart
3. Optionally uninstall the old demo from the Admin tool, and then restore the OPENC3_DEMO line in your .env

MTI-GMays, 2022-12-05:
There were two issues run issues in the init container:
1. I had a syntax issue on the API file I created
2. Also on line 594 of openc3\bin\openc3cli the `client.create(ENV[])` (screenshot attached). This happened after adding a new bucket to '.env'. I see that the openc3cli file has been changed in the current repository. 

[image]

ryanmelt, 2022-12-07:
Closing as OBE.

closed 2022-12-07 by ryanmelt