OpenC3/cosmos #304: OPENC3_CLOUD environment variable is required (RuntimeError), when building Bridge support in OpenC3\COSMOS

jurgensymynck, 2022-12-02:
**Describe the bug**
See issue, https://github.com/OpenC3/cosmos/issues/285 that deals with the exact same error, but for the `OpenC3/COSMOS-project` way of installing building and running COSMOS, not the`OpenC3/COSMOS` way as is the case here:

**To Reproduce**
1. start with clean install of OpenC3/COSMOS (not OpenC3/COSMOS-project) in folder c:\OpenC3
2. using git, reset master branch to tag '5.1.1', the latest stable version at this time. 
3. edit .env, remove OPENC3_DEMO line, set OPENC3_TAG=5.1.1. Leave rest as is.
4. Proceed to steps to build Bridge support. Super abriged version below, relevant tickets here: https://github.com/BallAerospace/COSMOS/issues/1508 
   - Requires a host Ruby installation (Ruby 3) including the devkit!
   - set RUBY.exe in system path? PATH = …;c:\ruby\bin;…
   - set host environment variable RUBYLIB= C:\openc3\openc3\lib
   - set host environment variable RUBYGEMS_URL=https://rubygems.org
   - reboot computer to activate environment variables
   - cd C:\OpenC3\openc3
   - bundle install 
   - cd ..
   - ruby.exe .\openc3\bin\openc3cli bridgesetup
          C:/openc3/openc3/lib/openc3/utilities/bucket.rb:23:in `getClient': OPENC3_CLOUD environment variable is required (RuntimeError) from ./openc3/bin/openc3cli:47:in `<main>' 

The previous solution was using an updated OpenC3.bat
Can I safely use the openc3.bat from `OpenC3/COSMOS-project` version 5.1.1 (latest stable) to continue or does this open a new can of worms?

**Environment (please complete the following information):**
- Windows 10
- OpenC3 COSMOS should be 5.1.1
   - from localhost:2900 "OpenC3 COSMOS Unknown © 2022 - License: AGPLv3"
   - git hash used: f5192ade6684d3691daf022b079646820f847ce6
     Author: ryanmelt <ryanmelt@users.noreply.github.com>  2022-11-19 04:19:28
     Tags: v5.1.1  
- OpenC3 COSMOS-project 5.1.1

ryanmelt, 2022-12-07:
Will better document this and probably make OPENC3_CLOUD have a default value so it will always run for bridges.

jmthomas, 2022-12-07:
Create an environment variable called OPENC3_CLOUD and set it to ‘local’.

Nileshkartik95, 2022-12-08:
Thanks Jason for your response!

I have tried setting the environment variable OPENC3_CLOUD to local but I came across an error with the local_bucket.rb file, although  local_bucket.rb is present in the ...cosmos\openc3\lib\openc3\utilities
as mentioned in this error. Just curious to know what could have gone wrong, is it interpreting this file with a different name 'local_' _bucket.rb seems something of this kind.
Please provide your feedback on the topic.

Following is the error message while executing cmd: ruby cosmos/openc3/bin/openc3cli
**Error Log:** 
{"time":1670462004583888600,"@timestamp":"2022-12-07T18:13:24.583-07:00","severity":"ERROR","container_name":"DESKTOP-1C74LAT","log":"Unable to require openc3/utilities/'local_'_bucket.rb due to cannot load such file -- openc3/utilities/'local_'_bucket.rb\nDid you mean?  openc3/utilities/local_bucket. Ensure openc3/utilities/'local_'_bucket.rb is in the OpenC3 lib directory."}
C:/Ruby31-x64/lib/ruby/3.1.0/socket.rb:61:in `connect_internal': A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond. - user specified timeout (Redis::CannotConnectError)

Thanks for your support!

jmthomas, 2022-12-08:
On unix: `export OPENC3_CLOUD=local`
On Windows: `set OPENC3_CLOUD=local`

No quotes around local.

closed 2022-12-09 by jmthomas

jurgensymynck, 2023-01-07:
Thanks a lot!

I am finally up and running! 
- Arduino Mega 2560 board 
- using serial connection over USB...
- Windows 10
- COSMOS 5.2.0 with bridge support

greetings, Jurgen

PR #320 (2022-12-09): Set OPENC3_CLOUD default as local
closes #304