OpenC3/cosmos #328: Help with JSONAccessor for TLM

bradenvirdellgenevatech, 2022-12-12:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

I am trying to use the JsonAccessor on a TLM packet. I cant seem to figure out what I am meant to send to the TLM from the interface. So far I have tried the standard binary packet and a JsonPacket. The JsonPacket seems promising but it runs into an issue where it doesnt have a packet.received_time as a method and thus can not continue.

I am currently not quite sure what I am missing

ryanmelt, 2022-12-13:
Interfaces always return Packet objects (Not JsonPacket).  For the JsonAccessor, the packet.buffer should be the JSON text.

bradenvirdellgenevatech, 2022-12-13:
Got it working, looks like I had to directly change the @accessor from BinaryAcessor to JsonAccessor in the protocol of the packet before sending it.

closed 2022-12-13 by bradenvirdellgenevatech