OpenC3/cosmos #354: ERROR MicroserviceOperator: MissingCredentialsError

Nileshkartik95, 2023-01-05:
Hello,

We followed the bridge solution provided in the following link:
>> [Create documentation for bridges (serial ports) · Issue #16 · OpenC3/openc3.com (github.com)](https://github.com/OpenC3/openc3.com/issues/16)

We are currently using the Windows operating system, where we are trying to implement the bridge solution.

We made the following changes to the bridge file:

## cosmos plugin interface:
INTERFACE <%= steam_target_name %>_INT tcpip_client_interface.rb COM21 COM21 10.0 10.0 LENGTH 32 16 17 1 BIG_ENDIAN 

after which we are facing the following issue,

`Unexpected process died... respawning! ruby plugin_microservice.rb
Stdout:
{"time":1672877273126394730,"@timestamp":"2023-01-05T00:07:53.126+00:00","severity":"INFO","microservice_name":"DEFAULT__INTERFACE__STEAM2_INT","container_name":"bc7ac9a8d59d","log":"Microservice initialized with config:
{\"name\"=>\"DEFAULT__INTERFACE__STEAM2_INT\", \"folder_name\"=>nil, \"cmd\"=>[\"ruby\", \"interface_microservice.rb\", \"DEFAULT__INTERFACE__STEAM2_INT\"], \"work_dir\"=>\"/openc3/lib/openc3/microservices\", \"ports\"=>[], \"env\"=>{}, \"topics\"=>[], \"target_names\"=>[\"STEAM2\"], \"options\"=>[], \"container\"=>nil, \"updated_at\"=>1672877261965476398, \"plugin\"=>\"openc3-steam2-1.1.2.20221209010541.gem__20230104011137\", \"needs_dependencies\"=>false}"}

Stderr:
/usr/lib/ruby/gems/3.1.0/gems/aws-sigv4-1.5.2/lib/aws-sigv4/signer.rb:654:in `extract_credentials_provider': missing credentials, provide credentials with one of the following options: (Aws::Sigv4::Errors::MissingCredentialsError)
  - :access_key_id and :secret_access_key
  - :credentials
  - :credentials_provider
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sigv4-1.5.2/lib/aws-sigv4/signer.rb:144:in `initialize'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:235:in `new'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:235:in `build_v4_signer'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:14:in `block in <class:S3Signer>'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:72:in `call'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:219:in `block in resolve_defaults'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:59:in `each'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:59:in `each'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:218:in `resolve_defaults'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:211:in `value_at'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:191:in `block in resolve'
    from /usr/lib/ruby/3.1.0/set.rb:511:in `each_key'
    from /usr/lib/ruby/3.1.0/set.rb:511:in `each'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:191:in `resolve'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:179:in `apply_defaults'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/configuration.rb:152:in `build!'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/base.rb:65:in `build_config'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/base.rb:22:in `initialize'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/client.rb:435:in `initialize'
    from /usr/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.158.0/lib/seahorse/client/base.rb:102:in `new'
    from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.0.11/lib/openc3/microservices/microservice.rb:125:in `initialize'
    from plugin_microservice.rb:27:in `initialize'
    from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.0.11/lib/openc3/microservices/microservice.rb:43:in `new'
    from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.0.11/lib/openc3/microservices/microservice.rb:43:in `run'
    from plugin_microservice.rb:43:in `<main>'`
[image]

I would highly appreciate your input on the following issue.

jmthomas, 2023-02-20:
@Nileshkartik95 can you try this on the latest release and let us know if this is still an issue?

ryanmelt, 2023-02-21:
Your interface line is wrong:
```
INTERFACE <%= steam_target_name %>_INT tcpip_client_interface.rb COM21 COM21 10.0 10.0 LENGTH 32 16 17 1 BIG_ENDIAN 
```

You've got serial port COM21 in there for some reason.   It should have host.docker.internal and a couple TCPIP ports.   I'll see if I get the same errors if I do what you did.

ryanmelt, 2023-02-21:
I get completely different errors.  Please share your bridge.txt and plugin.txt

PR #504 (2023-02-22): Bridge updates for bridgegem system
closes #497 
closes #354 
closes #303 

Example bridges:
https://github.com/OpenC3/openc3-cosmos-bridge-serial
https://github.com/OpenC3/openc3-cosmos-bridge-hidapi
https://github.com/OpenC3/openc3-cosmos-bridge-dualsense

Bridge documentation: https://openc3.com/docs/v5/bridges
closed 2023-02-22 by ryanmelt