OpenC3/cosmos #355: Create ArrayPlot Widget

ryanmelt, 2023-01-05:
Create a new TlmViewer Widget that takes an array item and plots all of its values in a graph.
Example use: Quick plot of FFT data

ryanmelt, 2023-01-18:
Add bar chart option

skliper, 2023-08-29:
Is it possible to graph array values currently?  Either all or one element?  Whenever I select one the scale goes screwy and I don't see how to select just one element.

[image]

jmthomas, 2023-11-14:
@skliper we implemented the graphing of individual array items in #933

ryanmelt, 2024-12-13:
Also support feature for a fixed user provided line that is overlayed and used as a comparison/reference for the data.

jmthomas, 2025-01-18:
We already support arbitrary horizontal lines on a graph.

jmthomas, 2025-01-28:
Given an array of Y values, create the following:
[image]

ryanmelt, 2025-01-29:
I would assume an array of x,y pairs rather than an interval.

PR #1868 (2025-01-30): Arrayplot widget
closes #355
closed 2025-01-30 by jmthomas