OpenC3/cosmos #356: OpenC3Api class is failing to send scope as kwparam

MTI-twalker, 2023-01-06:
Here is the log of the request I am sending:
In my tool TOOLNAME.vue I have imported OpenC3Api

```
import { OpenC3Api } from '@openc3/tool-common/src/services/openc3-api'

...
    data() {
        return {
            api: new OpenC3Api(),
        }
     }
...
this.api.get_target(this.targetName).then((target) => {
    //Do Something with target data
}
```

The tool did work in prior versions (I will return to this thought later). The log from the cmd_tlm_api container shows the request filled below and as you can see the keyword_params content is empty:
```
{
"time":1673038288923874307,"@timestamp":"2023-01-06T20:51:28.923+00:00","severity":"INFO",
        "microservice_name":"CMD__TLM__API",
        "user":"Unknown",
        "container_name":"1234",
        "log":"API 
        data: {
          \"jsonrpc\":\"2.0\",
          \"method\":\"get_target\",
          \"params\":[\"<TARGET_NAME>\"],
          \"id\":4,
          \"keyword_params\":{}
       }"
}

{
"time":1673038288924962527,"@timestamp":"2023-01-06T20:51:28.924+00:00","severity":"ERROR",
        "microservice_name":"CMD__TLM__API",
        "container_name":"1234",
           "log":"AuthError : Scope is required
           /usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/authorization.rb:41:
               in `authorize'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/api/target_api.rb:49:
               in `get_target'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/io/json_drb.rb:267:
               in `public_send'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/io/json_drb.rb:267:
               in `process_request'\n/src/app/controllers/api_controller.rb:87:
               in `handle_post'\n/src/app/controllers/api_controller.rb:41:
               in `block in api'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/open_telemetry.rb:52:
               in `in_span'\n/src/app/controllers/api_controller.rb:28:in `api'"
}

```

I found that this is auto-populated in the openc3-api.js Line 39. Which is loading:
`kwparams['scope'] = window.openc3Scope`

It appears that the scope is no longer stored in localStorage but is loaded in the window object somewhere. I am not entirely sure how this change works but it appears to work for everything that is using it except the way I am calling it, maybe?

This was discovered when a co-worker was getting kicked from what I believe was due to a 401 error response (due to no scope) which then called updateToken and cleared the localStorage of the token and then was redirected to a login screen. However, mine browser was loading it just fine. 

What I found is my browser still had the scope key loaded in localStorage with the value of DEFAULT. This was likely placed their with older versons of OpenC3 and just remained stored in the browser. I removed that key and did a hard reload on OpenC3 and the scope key never repopulates. Which then caused me to have issues loading my tool. So outside of when the update back in august to use window.openc3Scope instead of localStorage.scope, I am not sure when this issue likely showed up because I have always had that key stored in my browser.

So when I add back in the scope key to the browser, my request in the log looks like this:
```
{
"time":1673039535991795079,"@timestamp":"2023-01-06T21:12:15.991+00:00","severity":"INFO",
    "microservice_name":"CMD__TLM__API",
    "user":"Unknown",
    "container_name":"1234",
    "log":"API 
    data: {
        \"jsonrpc\":\"2.0\",
        \"method\":\"get_target\",
        \"params\":[\"<TARGET_NAME>\"],
        \"id\":2,
        \"keyword_params\":{\"scope\":\"DEFAULT\"}
    }"
}

Completed 200 OK in 2ms (Allocations: 984)
```

Everything else appears to work without that scope key except my tool and from what I can tell this is the cause. I looked at some of the other tools code and I don't see where I am doing anything different. 

Any insight? Similar experiences?

jmthomas, 2023-01-09:
I believe [this PR](https://github.com/OpenC3/cosmos/pull/324) was responsible which was in [COSMOS 5.2.0](https://github.com/OpenC3/cosmos/releases/tag/v5.2.0).

@ryanmelt?

ryanmelt, 2023-02-01:
It was actually this PR way back in August: https://github.com/OpenC3/cosmos/pull/115

ryanmelt, 2023-02-01:
I believe this was should have been fixed by updating the tool-common npm package. 
See https://github.com/OpenC3/cosmos/issues/362
Make sure your package.json references a more recent version of tool-common, and I think that should fix your tool.

closed 2023-02-01 by ryanmelt