OpenC3/cosmos #358: CmdTlmServer doesn't recognize individual, well separated serial packets?

jurgensymynck, 2023-01-07:
Hello,

I am wondering what I could have done wrong: I receive serial packets from (on target TISCO, called TLM_PING) that are spaced apart 10 ms, but still COSMOS interpretes these more often than not as one packet, resulting in a log message like:
` ERROR	DEFAULT__INTERFACE__TISCO_INT	TISCO TLM_PING received with actual packet length of 18 but defined length of 9 `
Some packets get registered correctly.

Any idea's? Is there something missing in my interface definitions?
Thanks in advance, Jurgen

## Background info:
### Oscilloscope images
... with at the bottom the actual content of a typical TLM_PING frame.
In this case: 0x55, 0x55, 0xD5 (a Sync pattern), 0x81 (packet ID), 0x9 (frame size), followed by four bytes representing an acquisition time stamp.
[image]
[image]

### A faulty raw packet
... from CmdTlmServer. Note the timestamp bytes are different, the screenshot was taken at different time.
```
Raw Telemetry Packet: TISCO TLM_PING
Received Time:
Count: 53580
Address   Data                                             Ascii
---------------------------------------------------------------------------
00000000: 55 55 d5 81 09 43 42 0c 00 55 55 d5 81 09 4d 42  UU   CB  UU   MB
00000010: 0c 00
```
### Telemetry definition text
```
TELEMETRY TISCO TLM_PING LITTLE_ENDIAN
  APPEND_ITEM SYNC_PATTERN 24 BLOCK "Sync Pattern"
  APPEND_ID_ITEM PACKET_ID 8 UINT 0x81 "Packet ID"
  APPEND_ITEM FRAME_SIZE 8 UINT "Frame size"
    UNITS Bytes B
  APPEND_ITEM TIMESTAMP 32 UINT "TIMESTAMP since Arduino startup (millis())"
    UNITS milliseconds ms
```
### plugin.txt
```
VARIABLE tisco_target_name TISCO
TARGET TISCO <%= tisco_target_name %>
INTERFACE  <%= tisco_target_name %>_INT tcpip_client_interface.rb host.docker.internal 2950 2950 10.0 nil BURST
  MAP_TARGET <%= tisco_target_name %>
```
### bridge.txt
```
INTERFACE SERIAL_INT serial_interface.rb COM5 COM5 115200 NONE 1 10.0 nil LENGTH 32 8 0 1 LITTLE_ENDIAN 0 0x5555D5 nil true nil
ROUTER SERIAL_ROUTER tcpip_server_interface.rb 2950 2950 10.0 nil BURST
    ROUTE SERIAL_INT
```
### COSMOS version
from localhost:2900: OpenC3 COSMOS (5.2.0) © 2022 - License: AGPLv3
from Git:
```
Sha1 ID: 59887253fd1f78bde363207b71f946d5f366b57d
Author: ryanmelt <ryanmelt@users.noreply.github.com>  2022-12-14 00:02:41
Committer: github-actions[bot] <41898282+github-actions[bot]@users.noreply.github.com>  2022-12-14 00:02:41
Tags: v5.2.0
Branches: main, remotes/origin/main
    [Github Action] Update version to 5.2.0
```

PratikAher20, 2023-01-20:
Hi @jurgensymynck , I am beginner to use COSMOS 5. Can you just tell me in which directory did you place that bridge.txt file? And did you include it anywhere?

jurgensymynck, 2023-01-20:
Hi, @PratikAher20 

The `ruby.exe .\openc3\bin\openc3cli bridgesetup` dumps a template `bridge.txt`config file in the current folder. In the following case, that's 'C:\OpenC3\`.
The location to the bridge.txt file doesn't matter, as long as you run the subsequent commands correctly.

Check my other bug report in ticket https://github.com/OpenC3/cosmos/issues/304 . This bug should be solved. Proceed from `ruby.exe .\openc3\bin\openc3cli bridgesetup`

* `cd C:\OpenC3\`
* `ruby.exe .\openc3\bin\openc3cli bridgesetup`

Then I moved my generated `bridge.txt` to the root of my project folder (`C:\COSMOS\TiSCo`) where my Arduino code resides, to have it included in the Git commit history. There, modify the bridge configuration file to your needs (change serial port settings).

* go to your project folder : `cd  C:\COSMOS\TiSCo` 
* Start the bridge: run `ruby.exe C:\OpenC3\openc3\bin\openc3cli bridge bridge.txt` which runs the bridge setup using the `bridge.txt` config file , which resides in the same folder as where you ran `ruby.exe`.

Good luck!

PratikAher20, 2023-01-22:
Thanks @jurgensymynck it worked for me..

ryanmelt, 2023-02-01:
Your plugin.txt should be using the LENGTH protocol not burst.  Something like this:

```
VARIABLE tisco_target_name TISCO
TARGET TISCO <%= tisco_target_name %>
INTERFACE  <%= tisco_target_name %>_INT tcpip_client_interface.rb host.docker.internal 2950 2950 10.0 nil LENGTH 32 8  0 1 LITTLE_ENDIAN 0 0x5555D5
  MAP_TARGET <%= tisco_target_name %>
```

closed 2023-02-01 by ryanmelt