OpenC3/cosmos #359: Is there anywhere in documentation that states why and when the Override Protocol was deprecated?

GTI-Dakota, 2023-01-09:
Updated to OpenC3 5.3.0 and a couple interfaces are now crashing. Turns out the Override Protocol is deprecated. In your release notes, it mentions that the override protocol is no longer necessary thanks to the overriding telemetry, but I only found out that the protocol file itself was deprecated once the update had executed.

Thanks,

Dakota

jmthomas, 2023-01-10:
I left a file called override_protocol.rb in the COSMOS filesystem so any require would still be successful. I thought that would be enough since it no longer is necessary to override telemetry. I guess I should have left a skeleton class or something in case someone was actually using the class. Were you using it (inheriting from it?) or just requiring it?

GTI-Dakota, 2023-01-11:
We were using it for a couple interfaces. We have since updated them to deprecate the protocol, but initially on the update a few of our interfaces and targets were unusable without a heads-up

closed 2023-01-11 by GTI-Dakota