OpenC3/cosmos #361: Web-Application Prefix

dmitri-mcguckin, 2023-01-09:
Hello,

I'd like to request the feature of inserting web-application prefixes into HTTP requests/responses for COSMOS.

I.e. being able to add arbitrary prefixes such as `/cosmos-dev-1` and `/cosmos-dev-2`.

This is particularly helpful when we have one server with multiple services, or the more-likely case of multiple instances of the same service all on one machine.

&nbsp;

***

**Is your feature request related to a problem? Please describe.**

I was experimenting with some work-arounds to the root-level app and when setting up a proper NGinx Reverse-Proxy  to remove a prefix, *(`/cosmos-dev` for example)*, I can see from the docker logs that the routing does infact make it to the web application with the correct route. When opening it in the browser, however, all of the response-paths still assume a `/` prefix and thus 404 in the browser, as seen in the image below.

As a temporary work-around, we have internal subdomains setup to deal with routing, however, the ideal case would be to have only one subdomain for our test server.

```
172.24.0.1 - - [09/Jan/2023:17:04:46 +0000] "GET /login HTTP/1.0" 200 2476 "-" "-" 1645 "web-router@file" "http://openc3-minio:9000" 7ms
172.24.0.1 - - [09/Jan/2023:17:04:59 +0000] "GET /login HTTP/1.0" 200 2476 "-" "-" 1646 "web-router@file" "http://openc3-minio:9000" 1ms
172.24.0.1 - - [09/Jan/2023:17:05:04 +0000] "GET /login HTTP/1.0" 200 2476 "-" "-" 1647 "web-router@file" "http://openc3-minio:9000" 7ms
172.24.0.1 - - [09/Jan/2023:17:05:08 +0000] "GET /login HTTP/1.0" 200 2476 "-" "-" 1648 "web-router@file" "http://openc3-minio:9000" 7ms
172.24.0.1 - - [09/Jan/2023:17:05:10 +0000] "GET /login HTTP/1.0" 200 2476 "-" "-" 1649 "web-router@file" "http://openc3-minio:9000" 7ms
```

[image]

&nbsp;

***

**Describe the solution you'd like**
I would like the ability to insert arbitrary prefixes to the web-application service, *(such as `/cosmos-dev-1`)*, such that the WSGI functions correctly.

&nbsp;

***

Please let me know if this already exists somewhere or of additional context is needed.

All the best.

jmthomas, 2023-01-10:
In the [cosmos-project](https://github.com/OpenC3/cosmos-project) our [compose.yaml](https://github.com/OpenC3/cosmos-project/blob/main/compose.yaml) file loads in a traefik configuration for the openc3-traefik service. The default [traefik.yaml](https://github.com/OpenC3/cosmos-project/blob/main/openc3-traefik/traefik.yaml) file is where we configure our routing. I think you could add various paths to this to direct back to our services.

dmitri-mcguckin, 2023-01-10:
I will check that out, thanks!

closed 2023-01-30 by jmthomas