OpenC3/cosmos #362: @openc3/tool-common needs update

MTI-GMays, 2023-01-09:
Using custom tools in OpenC3 5.3.0 will result in continuous log out (401 issues). The tool would previously work in OpenC3 5.1.1. The npm/@openc3/tools-common is out of date. once the '\node_modules\@openc3\tool-common\src\services\openc3-api.js' is updated to the what 'cosmos-5.3.0\openc3-cosmos-init\plugins\packages\openc3-tool-common\src\services\openc3-api.js' has, the issue is resolved. Ticket #356 is having this issue. 

The latest npm/@openc3/tool-common is 5.0.6 while OpenC3 release 5.3.0 is on @openc3/tool-common 5.3.0.
https://socket.dev/npm/package/@openc3/tool-common

[image]

jmthomas, 2023-01-10:
Looks like we need to set the correct tag in NPM. Thanks @MTI-GMays!

ryanmelt, 2023-01-12:
Fixed the tag on the 5.3.0 to latest. Also update the release process to tag correctly going forward.  
closed by #381

MTI-GMays, 2023-01-12:
Thanks Ryan!

PR #381 (2023-01-12): Fix github release if checks
closes #362 
See: https://github.com/actions/runner/issues/1483
closed 2023-01-12 by ryanmelt