OpenC3/cosmos #400: Error with mkdir of targets_modified in plugins folder on clean install

mdicicco, 2023-01-24:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
New user. Tried to do a clean install following the tutorial instructions and the openc3-cosmos-init-1 process crashes with an error about dir_s_mkdir failing with permission denied.  I saw a similar bug in issue #11 but the solution of fixing the git config linefeed did not solve the problem.  I did not add the `--user root` as I'm not running cygwin. 
```
2023-01-21 19:48:58 /usr/lib/ruby/3.1.0/fileutils.rb:243:in `mkdir': Permission denied @ dir_s_mkdir - /plugins/DEFAULT/targets_modified (Errno::EACCES)
2023-01-21 19:48:58     from /usr/lib/ruby/3.1.0/fileutils.rb:243:in `fu_mkdir'
2023-01-21 19:48:58     from /usr/lib/ruby/3.1.0/fileutils.rb:221:in `block (2 levels) in mkdir_p'
2023-01-21 19:48:58     from /usr/lib/ruby/3.1.0/fileutils.rb:219:in `reverse_each'
2023-01-21 19:48:58     from /usr/lib/ruby/3.1.0/fileutils.rb:219:in `block in mkdir_p'
2023-01-21 19:48:58     from /usr/lib/ruby/3.1.0/fileutils.rb:211:in `each'
2023-01-21 19:48:58     from /usr/lib/ruby/3.1.0/fileutils.rb:211:in `mkdir_p'
2023-01-21 19:48:58     from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/local_mode.rb:450:in `build_local_catalog'
2023-01-21 19:48:58     from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/local_mode.rb:476:in `sync_with_bucket'
2023-01-21 19:48:58     from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/local_mode.rb:328:in `block in sync_targets_modified'
2023-01-21 19:48:58     from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/local_mode.rb:327:in `each'
2023-01-21 19:48:58     from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/local_mode.rb:327:in `sync_targets_modified'
2023-01-21 19:48:58     from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.3.0/lib/openc3/utilities/local_mode.rb:80:in `local_init'
2023-01-21 19:48:58     from /openc3/bin/openc3cli:624:in `<main>'
```

**To Reproduce**
1. Follow instructions on wiki for installation
2. run with openc3.sh start
3. observe the failure in the open3-cosmos-init-1 console within docker desktop

**Expected behavior**
init process crashes and restarts repeatedly.  Browser won't connect.

**Screenshots**
N/A

**Environment (please complete the following information):**

- OS: Xubuntu 20.04
- OpenC3 COSMOS Version 5.3.0
- Browser
  -- [] Internet Explorer (IE)
  -- [] Google Chrome
  -- [X] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

mdicicco, 2023-01-27:
manually adding the 'targets_modified' folder to my plugins folder fixed the problem.  that's probably not the ideal fix but I felt it was worth mentioning

jmthomas, 2023-01-28:
What user are you running as? `whoami`
What are the directory permissions where you checked out COSMOS? `ls -la`

mdicicco, 2023-01-29:
I'm running as my normal user.  I made the repo in my home folder so nothing unique on the permissions.  The folder is `drwxrwxr-x` when viewed from my home dir an the plugins folder inside of the checked out repo is also `drwxrwxr-x`

ryanmelt, 2023-02-01:
This should be addressed by https://github.com/OpenC3/cosmos/pull/437

closed 2023-02-01 by ryanmelt