OpenC3/cosmos #402: Usage of received_count when parsing stored telemetry

jeffstjean, 2023-01-24:
I believe this question is closely related to #197 but I'm not sure if the PR and associated issues (#183, #196) fully answer my question.

We have files that are downloaded and parsed using a microservice. Some of these files may contain one or more telemetry packets which we wish to insert into the database as stored telemetry. I am unsure how to properly set the received_count for these packets. Should it be one greater than what's stored in the CVT?

The routine currently looks like this:
```
data = "some data" # packed data
packet = System.telemetry.identify_and_define_packet(Packet.new(nil, nil, :BIG_ENDIAN, nil, data))
raise("Unable to identify packet") if packet == nil

packet.stored = true
packet.received_count = 0
packet.received_time = time # from elsewhere

topic = "#{@scope)}__TELEMETRY__{#{@target}}__#{@name}"
msg_id, msg_hash = Topic.get_newest_message(topic)
packet.received_count = packet["received_count"] if msg_hash

TelemetryTopic.write_packet(packet, scope: download.scope())
```

Is this the right topic to grab the received_count from? Is received_count just supposed to represent the absolute count when that packet was received (ie. if these don't increment by one for each new packet, is that ok?)?

ryanmelt, 2023-01-25:
Received count was never meant be absolute.  It was intended to just be an incrementing number so you can differentiate packets.  However, we have made changes recently such that if an Interface microservice resets, the received count keeps going where it left off.

Received count is not stored in the raw log files, so historically when reading log files, we have just set it to 1, 2, 3, etc. as we read through the file for each packet.

jeffstjean, 2023-01-26:
So if I set the received count to anything (even zero) it shouldn't have any consequences? Or should I try and stay in sync with the count from the telemetry topic?

jmthomas, 2023-01-27:
We set received_count to zero in other places and return 0 if it is not set so that is fine. Received count doesn't really make sense with stored telemetry because it's coming in outside of the real time processing.

closed 2023-01-27 by jmthomas