OpenC3/cosmos #403: Associated Telemetry Points with Commands

ryanmelt, 2023-01-25:
Add command keywords to reference telemetry points.
This will allow us to show associated telemetry in CmdSender.

ryanbarna, 2023-03-24:
I can make this its own ticket if preferred, but this seemed like a similar enough enhancement that I'll start by commenting here. 

I am working on a similar feature for a small satellite ground station and would love to see this implemented. When I request single parameters from the satellite it records a `transaction ID` field from the request I made and fills a corresponding `transaction ID` field in its response. Each new command gets it own `transaction ID` until the maximum value of this bitfield is reached, causing the counter to reset. 

These transaction IDs would make it easy to associate corresponding commands and telemetry, but I cannot find a way to dynamically change the `ID_ITEM` in a COSMOS telemetry packet. My plan was to create a custom protocol to store a list of key-value pairs for the command names and their `transaction IDs`. When I received a packet, I would search this list for the corresponding `transaction ID` and associate the match with a similarly named telemetry point. However, I have had no success since the `ID_ITEMS` seem immutable once the plugin is deployed, making it difficult to do any kind of dynamic association. I have tried forcing the packet identification and adding my own functions to the `System.telemetry` class with no change in outcome. 

I am running out of time to develop this feature, but I would appreciate any suggestions you might have.

ryanbarna, 2023-03-28:
@ryanmelt Good to know. I'll take a look.

PR #1005 (2024-01-07): New cmd keywords for related telemetry and remove legacy snmp
closes #403
closed 2024-01-07 by ryanmelt