OpenC3/cosmos #419: Dynamically declared telemetry packets

jmthomas, 2023-01-29:
During a recent training we discussed supporting dynamically declared packets. Conops is that various commands are sent to the target embedded software and the contents of a packet changes. COSMOS only updates packet definitions based on plugins so this would require building and installing a new plugin. However you can break up plugins however you want so you can create two plugins: one for the statically defined packets and one that changes based on target SW updates.

ShaurCapella, 2023-03-11:
One of the things that we've done internally for COSMOS 3/4 design is have a custom config directory that we use to overwrite  core configs w/ soft links before the sw startup for the non-web-based cosmos design. This means we can overload a target file (connect to different ports depending on the asset) or even load different dictionaries (target cmd tlm definitions).

Given the way plugins were explained to me in  issue 539, i wonder if adding a feature button that allows one to rebuild and reload plugins at runtime based on a new config would solve the issue. For example in the default scope plugin folder you can have a directory like ```targets_modified``` but instead it's ```targets_redefined``` where say in the base plugin you have the target ```SPACECRAFT``` and if you include that same folder in ```targets_redefined``` you can actually rebuild the base plugin with all the same configs minus the ```SPACECRAFT``` folder, and same for the plugin.txt file, you can redefine ports or configs or targets as needed (this situation is for where you have a test asset but are adding and removing hardware ```

An example folder structure could be 
```
PLUGINS/
├─ DEFAULT/
│  ├─ targets_redefined/
│  │  ├─ plugin.txt
│  │  ├─ SPACECRAFT/
│  │  │  ├─ cmd_tlm.txt
```

ShaurCapella, 2023-03-18:
Thought I'd see if you had any thoughts on this! I can look into producing this feature and creating a PR if it's of interest.

PR #1048 (2024-01-18): Dynamic target packet definition creation support. 
Also 
Improve build time to only rebuild docs tools on base change
Update docusaurus

closes #419 
closes #1026 
closes #1044
closed 2024-01-18 by ryanmelt