OpenC3/cosmos #420: Plugin-Level REQUIRE for common target routines

jeffstjean, 2023-01-29:
The `REQUIRE` documentation says that the target's lib directory will be checked first, followed by the openc3 lib directory. We have some routines that are common across targets in the same plugin and would like to only have one instance of the ruby file present. Could this be modified to check a lib directory in the plugin directory before checking the openc3 directory? 

I think a potential solution might look like this in `target.rb`
```
when 'REQUIRE'
  usage = "#{keyword} <FILENAME>"
  parser.verify_num_parameters(1, 1, usage)
  filename = File.join(@dir, 'lib', parameters[0])
  begin
    # Require absolute path to file in target lib folder. Prevents name
    # conflicts at the require step
    OpenC3.disable_warnings do
      OpenC3.require_file(filename, false)
    end
  rescue LoadError
    begin
      filename = File.join(@dir, '../lib', parameters[0])
      OpenC3.disable_warnings do
        OpenC3.require_file(filename, false)
      end
    rescue LoadError
      begin
        # If we couldn't load at the target/lib level check everywhere
        OpenC3.disable_warnings do
          filename = parameters[0]
          OpenC3.require_file(parameters[0])
        end
      rescue Exception => err
        raise parser.error(err.message)
      end
````

ryanmelt, 2023-02-01:
If you have a library that is needed by multiple targets, that should be placed in the top-level lib folder for the plugin (just like a normal ruby gem lib folder).   In that location a single file will work for all targets.

MTI-twalker, 2023-02-02:
I tested this by taking a working plugin I have that copies the files to each target. I moved them to a parent lib directory and removed the copies.
Adding a lib directory to the parent plugin directory does not appear to work. Is there a way to reference the require? 
These attempts at adding the require to the target.txt have all failed
```
REQUIRE lib/filename.rb
REQUIRE ../../lib/filename.rb
REQUIRE filename
require 'filename.rb' -> 
require_relative '../../lib/filename.rb' (require relative isnt recognized)
require '../../lib/filename.rb'
Not including any require/REQUIRE
```
Most times the crash message states:
 Unable to require lib/filename.rb due to cannot load such file -- lib/filename.rb. Ensure lib/filename.rb is in the OpenC3 lib directory.

I see that the gmspec file includes the lib directory as part of it but I clearly am not referencing this correct. 

I also added the require "filename.rb" to the top of each cmd_tlm definition that used it and it was unable to locate the file. 

```
<%
   require 'filename.rb'
%>
```
For sanity's sake, i moved the files back to each target's lib directory and the plugin loaded.

MTI-twalker, 2023-02-02:
All,
I managed to find a middle ground and it was hiding in details about the files that I didn't include as you would expect.

There are a couple notes:
I am not able to reference the script in the plugin.txt file unless that script was within the TARGET/lib directory.
- In this case an interface script

I am not able to reference a script in the common lib for a READ_CONVERSION

I am able to reference the script with the ERB if the functions were being used to render cmd_tlm definitions

jmthomas, 2023-02-20:
I was able to create a new plugin with a lib folder. I successfully referenced the top level lib from another script in my target's lib folder by simply requiring it `require 'top_script.rb'` where top_script.rb was at the top level lib folder. Same with a read conversion `READ_CONVERSION string_conversion.rb` where string_conversion.rb was at the top level lib folder.

I think some of the confusion was related to #456 and how we pull in dependencies.

closed 2023-02-20 by jmthomas