OpenC3/cosmos #425: Command Sequence tool for V5

MTI-twalker, 2023-01-31:
Are there any plans on porting the command sequence tool over to V5 from V4?

jmthomas, 2023-01-31:
It never felt like Command Sequence was that well used. With the Calendar tool we have the ability to execute arbitrary scripts or send commands at future times. That was the original goal of Command Sender: to execute a block of commands with various delays to sort of simulate a time based command sequence. I think Calendar makes Command Sender OBE.

MTI-twalker, 2023-01-31:
A team here uses the command sequence to for its export capabilities to generate ATS/RTS tables.

ryanmelt, 2023-02-01:
We'll add this to our roadmap.  It is useful for building flight software command tables.   Will probably be several months though.

jmthomas, 2023-02-06:
This could be a part of Table Manager if it understood the command definitions.

closed 2023-07-06 by MTI-twalker