OpenC3/cosmos #444: Uploading a plugin with a microservice causes all plugins to reinstall

MTI-twalker, 2023-02-02:
I am currently using 3 plugins that are unrelated (no shared targets, definitions, scripts) to a 4th plugin that is being worked in. 

The 4th plugin recently had a microservice added to it. Even if the other 3 plugins are fully installed, when the 4th is uploaded/installed it causes all interfaces to close and all plugins to reinstall.

jmthomas, 2023-02-03:
I was unable to reproduce. I ran our demo and then cloned the https://github.com/OpenC3/openc3-microservice-template and built `rake build VERSION=1.0.0` and uploaded. The demo interfaces were all fine and nothing was re-installed. Can you reproduce from the demo and the openc3-microservice-template?

MTI-twalker, 2023-02-03:
@jmthomas I was not able to recreate the issue with my local configuration either. There must have been something else. I went off some bad information as I was told that it was cause it had a microservice. Sorry about that!

MTI-twalker, 2023-02-04:
@jmthomas I was able to recreate (accidentally) the situation. 

This appears to happen when I am debugging an error.

To recreate:
- clone the microservice template
- Insert bug into microservice script:
- [image]
- build it: rake build VERSION=1.0.0
- Upload it
- Delete it

So the issue is actually when you remove a bugged microservice.

jmthomas, 2023-02-07:
@MTI-twalker I was unable to reproduce following your instructions. I had the demo running, cloned the openc3-microservice-template, `rake build VERSION=1.0.0`, installed, observed plugin failing:

```
Ruby microservice DEFAULT__USER__TEMPLATE file template_microservice.rb failed syntax check
template_microservice.rb:36: warning: found `= literal' in conditional, should be ==
template_microservice.rb:57: syntax error, unexpected end-of-input, expecting `end'
...roservice.run if __FILE__ == $0
...               
```

Went back to Admin and uninstalled plugin successfully. The demo was running happily the whole time.

MTI-twalker, 2023-03-14:
This was not able to be recreated with the latest version. This was an issue found in 5.3.0.