OpenC3/cosmos #448: Import COSMOS 4 Bin Files

tsjordan-eng, 2023-02-03:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

We have a lot of data collected with COSMOS 4 we'd like to be able to plot and review in COSMOS 5. 

I tried going to `localhost:2900/minio` on version 5.4.1 to try to upload my existing bin files but the screen is blank. This would be a solution, but there could also be an uploader/importer tool in the web interface. This wouldn't be done very often, so maybe it doesn't need to be as slick as that.

ryanmelt, 2023-02-03:
I'm going to come up with some sort of file dropbox concept, for processing logged data from any other source.

We need to eat up the log files, and then run them through the processing chain so that decom and reduction will occur.  

The COSMOS 4 log files are just the raw data, so you'll need to have a properly configured target in COSMOS 5 as well for this to work like you want.

ryanmelt, 2023-02-03:
Also thanks for pointing out that upgrading minio broke the minio console :(

tsjordan-eng, 2023-11-03:
Do you think I can use the COSMOS4 Replay feature and a COSMOS5 "Chain" interface to consume all my logs into COSMOS5?

ryanmelt, 2023-11-03:
That is an option that would probably work.  You would need to use the preidentified protocol with cosmos 4 mode.   Another option would be to create a direct COSMOS 4 file reading interface class.

jmthomas, 2025-07-22:
You can now do this with the [FileInterface](https://docs.openc3.com/docs/configuration/interfaces#file-interface) and the [PreidentifiedProtocol](https://docs.openc3.com/docs/configuration/protocols#preidentified-protocol)

closed 2025-07-22 by jmthomas