OpenC3/cosmos #456: Install ruby gems to all containers

MTI-twalker, 2023-02-04:
**Is your feature request related to a problem? Please describe.**
I have loaded a dependency to the gems through the Admin->GEMS. I confirmed in script runner that I can require it.
We are using that dependency to be used by a protocol script. When it loads my interface does not load due to "cannot load such file". 
I tried a few things and found my workaround after a bit of effort. I opened up the operator container terminal and I ran the "gem install <gem>" command in the lib directory.
After I did that I tried to load the plugin again and the interface successfully loaded.

**Describe the solution you'd like**
I would like the dependency gems to be loaded to all containers in order to not have to sign onto the operator container and separately install it.

ryanmelt, 2023-02-04:
Interfaces are isolated from gems unless I know they need them.  Try adding a top level lib folder to your plugin with at least one file in it, and then see if it works.

MTI-twalker, 2023-02-04:
That did it!

Ill leave this open due to the new title. Feel free to close it.

ryanmelt, 2023-02-04:
Good.   It tries to autodetect right now, but I should add a config option to force it.

PR #495 (2023-02-22): Add NEEDS_DEPENDENCIES keyword for plugins
closes #456
closed 2023-02-22 by jmthomas