OpenC3/cosmos #460: Incorrect CRC calculation when using LENGTH protocol

ryanbarna, 2023-02-06:
I am attempting to use the Length and CRC protocols to attach to the appropriate length field value and 16-bit CCITT-FALSE CRC to a CCSDS packet. However, it appears that the Length protocol may be called after the CRC protocol, causing the wrong CRC to be appended to the packet. 

For example, I get the correct CRC when I manually enter a `DATA_LEN` value of 7 and send this `EXAMPLE` packet using the Burst protocol.
```
COMMAND TEMP EXAMPLE  BIG_ENDIAN "Demonstration of CRC calculation"
  # Keyword           Name           BitSize Type   Min      Max Default
  APPEND_PARAMETER DATA_LEN 16 UINT 0x0 0xFFFF 0
  APPEND_PARAMETER DATA 32 UINT 0x0 0xFFFFFFFF 0
  APPEND_PARAMETER CRC 16 UINT 0x0 0xFFFF 0xFFFF
```
Below is the plugin.txt:
```
# Set VARIABLEs here to allow variation in your plugin
# See https://openc3.com/docs/v5/plugins for more information
VARIABLE temp_target_name TEMP

# Modify this according to your actual target connection
# See https://openc3.com/docs/v5/interfaces for more information
TARGET TEMP <%= temp_target_name %>
INTERFACE <%= temp_target_name %>_INT tcpip_client_interface.rb host.docker.internal 8080 8081 10.0 nil BURST
  MAP_TARGET <%= temp_target_name %> 
  PROTOCOL READ_WRITE CrcProtocol CRC false ERROR -16 16 BIG_ENDIAN nil nil nil nil nil
```
This will send the desired packet `00 07 00 00 00 00 69 c4`.

This is not the case when I change the plugin.txt to utilize the Length protocol. The new plugin.txt:
```
# Set VARIABLEs here to allow variation in your plugin
# See https://openc3.com/docs/v5/plugins for more information
VARIABLE temp_target_name TEMP

# Modify this according to your actual target connection
# See https://openc3.com/docs/v5/interfaces for more information
TARGET TEMP <%= temp_target_name %>
INTERFACE <%= temp_target_name %>_INT tcpip_client_interface.rb host.docker.internal 8080 8081 10.0 nil LENGTH 0 16 1 1 BIG_ENDIAN 0 nil nil true nil
  MAP_TARGET <%= temp_target_name %> 
  PROTOCOL READ_WRITE CrcProtocol CRC false ERROR -16 16 BIG_ENDIAN nil nil nil nil nil
```
Now, when I send the `EXAMPLE` packet COSMOS outputs the packet: `00 07 00 00 00 00 0e 10`. The Length protocol correctly fills the `DATA_LEN` field but appends the wrong CRC. Notice that this is the CRC that should be appended if the `DATA_LEN` value was its default `0x0000` value.

Version Information:
```
COSMOS: v5.4.2 
OS: Windows 10 Pro 22H2
Docker Engine: 20.10.21 
```

ryanmelt, 2023-02-06:
Try:

```
INTERFACE <%= temp_target_name %>_INT tcpip_client_interface.rb host.docker.internal 8080 8081 10.0 nil nil
  MAP_TARGET <%= temp_target_name %> 
  PROTOCOL READ_WRITE CrcProtocol CRC false ERROR -16 16 BIG_ENDIAN nil nil nil nil nil
  PROTOCOL READ_WRITE LengthProtocol 0 16 1 1 BIG_ENDIAN 0 nil nil true nil
```

ryanbarna, 2023-02-06:
It worked! Thanks for the help.

closed 2023-02-06 by ryanmelt