OpenC3/cosmos #461: Clients are always 0 after get_interface() in a microservice

GTI-Dakota, 2023-02-06:
**Describe the bug**
Client value is never updated past 0 when get_interface is called inside of a microservice. If clients are queried within a microservice, the CmdfTlmServer will never update the client count either so it will remain at 0.
This behavior is not seen if the same code runs in a script runner session.

**To Reproduce**
Steps to reproduce the behavior:

1. Create microservice to query and print an interface's info
`interface = get_interface("MY_INTERFACE_INT")`
`clients = interface['clients']`
`@logger.info("Clients: #{clients}")`

**Expected behavior**
Clients will remain at zero.

**Environment (please complete the following information):**

- OS: [Ubuntu 22.04]
- OpenC3 COSMOS Version: [5.4.2]
- Browser
  -- [x] Internet Explorer (IE)
  -- [x] Google Chrome
  -- [x] Chromium based browsers
  -- [x] Mozilla Firefox
  -- [] Safari

GTI-Dakota, 2023-02-06:
update: This seems to only occur when the interface is queried SOON after it first starts

jmthomas, 2023-03-06:
I was not able to reproduce. I created a plugin with a TCPIP server like so:
```
INTERFACE TEST_INT tcpip_server_interface.rb 7777 7777 10.0 nil BURST
  MAP_TARGET TEST
```
Also had to modify the compose.yaml to add the port:
```
  openc3-operator:
    ports:
      - "127.0.0.1:7777:7777"
```
Generated a microservice and added the following:
```
        interface = get_interface("TEST_INT")
        clients = interface['clients']
        Logger.info("Clients: #{clients}")
```
When I connected using curl like so `curl -v 127.0.0.l:7777 127.0.0.1:7777` I saw `Clients: 1`. Note that only a server has client counts so only our tcpip_server_interface.rb will show clients. Basically if you see clients in the Interfaces tab you should also see it using the `get_interface` api.

GTI-Dakota, 2023-03-07:
Thanks for taking a look. It's probable that my logic is flawed inside of my microservice then. I'll give it another pass.

Thanks!

closed 2023-03-07 by GTI-Dakota