OpenC3/cosmos #468: Allow users to turn off default tools

wiatrp, 2023-02-09:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We'd like a way to turn off default tools/plugins so COSMOS doesn't install the ones we aren't using by default. For example, we may not want the handbooks tool to be available.

**Describe the solution you'd like**
Allow the user to turn off tools similar to how the demo plugin is handled in `init.sh`. It would be nice to be able to set an env variable to turn off any of the tools individually.

[image]

**Additional context**
Another reason we would like to do this is because we will probably make heavy modifications to some tools. Instead of making those changes directly to source, we'd like to just create a new custom tool that mirrors a COSMOS one that we can modify instead. This makes upgrading the source easier. So, we may want to turn off the default calendar tool and create our own, but that means we don't want COSMOS to install the default calendar tool.

jmthomas, 2023-02-09:
I could see having something like:

```
if [ -z $OPENC3_NO_CALENDAR ]
    ruby /openc3/bin/openc3cli load /openc3/plugins/gems/openc3-cosmos-tool-calendar-*.gem || exit 1
fi
```

And it would install by default but you could add OPENC3_NO_XXX to your local .env file to prevent installation.

jmthomas, 2023-02-10:
closed by #473

closed 2023-02-10 by jmthomas