OpenC3/cosmos #474: Telemetry screen WIDTH setting not working in 5.4.2

djordan-nh, 2023-02-10:
**Describe the bug**
Telemetry screen widget WIDTH settings are not applying when displayed. Widgets are using their default widths instead.
This setting previously worked in 5.3.0, but stopped working when I upgraded to 5.4.2.

**To Reproduce**
Create a telemetry screen with custom widget WIDTH settings either using GLOBAL_SETTING or SETTING

**Expected behavior**
The widget widths are set to the defined value when displayed.

**Environment (please complete the following information):**

- OS: Windows 10
- OpenC3 COSMOS Version: 5.4.2
- Browser: Google Chrome

ryanmelt, 2023-02-14:
Can you be more specific.   Which widget isn't working?

jmthomas, 2023-02-15:
Note: WIDTH and HEIGHT now require units: https://openc3.com/docs/v5/telemetry-screens#setting. Please try with units and let me know if that is the issue.

jmthomas, 2023-02-15:
I think this is the issue but we're going to make no units default to 'px'. That way existing stuff doesn't break. Also the PADDING and MARGIN items currently don't respect units so I'll fix that as well. It's all in Widget.js.

closed 2023-02-16 by ryanmelt

djordan-nh, 2023-02-21:
Adding the px units to my WIDTH settings fixed it, thank you!

PR #484 (2023-02-16): Default various SETTINGS to px
closes #474