OpenC3/cosmos #497: MissingCredentialsError when I try to run bridge

DavidWathen, 2023-02-20:
Hi,

I'm running a local version of OpenC3 and when I try to run my bridge I get the following error:

/usr/local/lib/ruby/gems/3.1.0/gems/aws-sigv4-1.5.0/lib/aws-sigv4/signer.rb:646:in `extract_credentials_provider': missing credentials, provide credentials with one of the following options: (Aws::Sigv4::Errors::MissingCredentialsError)
  - :access_key_id and :secret_access_key
  - :credentials
  - :credentials_provider
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sigv4-1.5.0/lib/aws-sigv4/signer.rb:136:in `initialize'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:235:in `new'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:235:in `build_v4_signer'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:14:in `block in <class:S3Signer>'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:72:in `call'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:219:in `block in resolve_defaults'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:59:in `each'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:59:in `each'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:218:in `resolve_defaults'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:211:in `value_at'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:191:in `block in resolve'
	from /usr/local/lib/ruby/3.1.0/set.rb:511:in `each_key'
	from /usr/local/lib/ruby/3.1.0/set.rb:511:in `each'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:191:in `resolve'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:179:in `apply_defaults'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:152:in `build!'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/base.rb:65:in `build_config'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/base.rb:22:in `initialize'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/client.rb:435:in `initialize'
	from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/base.rb:102:in `new'

I don't understand why I would be getting any AWS error since I'm running local. Not from the Cloud.

Any help would be greatly appreciated!

David

jmthomas, 2023-02-20:
Seems related to #354

DavidWathen, 2023-02-20:
Yes I did see that issue but I couldn't find any resolution. 

Thanks for the clarification on the use of Ruby AWS library when talking to Minio. 

This is happening with 5.1.1 and the latest 5.4.2

I had a colleague check his machine and it is happening for him as well.

ryanmelt, 2023-02-21:
Are these errors on the COSMOS side, or on the host bridge side?

ryanmelt, 2023-02-21:
Can you share your plugin.txt INTERFACE line that connects to the bridge?

DavidWathen, 2023-02-21:
> [EXTERNAL EMAIL]DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe

Hi Ryan,

These are occurring on the bridge side when I try and start it.

/usr/local/bin/ruby ../../toolchain/openc3/openc3/bin/openc3cli bridge draco_cdh_bridge.txt
/usr/local/lib/ruby/gems/3.1.0/gems/aws-sigv4-1.5.0/lib/aws-sigv4/signer.rb:646:in `extract_credentials_provider': missing credentials, provide credentials with one of the following options: (Aws::Sigv4::Errors::MissingCredentialsError)
  - :access_key_id and :secret_access_key
  - :credentials
  - :credentials_provider
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sigv4-1.5.0/lib/aws-sigv4/signer.rb:136:in `initialize'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:235:in `new'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:235:in `build_v4_signer'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/plugins/s3_signer.rb:14:in `block in <class:S3Signer>'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:72:in `call'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:219:in `block in resolve_defaults'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:59:in `each'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:59:in `each'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:218:in `resolve_defaults'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:211:in `value_at'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:191:in `block in resolve'
               from /usr/local/lib/ruby/3.1.0/set.rb:511:in `each_key'
               from /usr/local/lib/ruby/3.1.0/set.rb:511:in `each'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:191:in `resolve'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:179:in `apply_defaults'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/configuration.rb:152:in `build!'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/base.rb:65:in `build_config'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/base.rb:22:in `initialize'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-s3-1.114.0/lib/aws-sdk-s3/client.rb:435:in `initialize'
               from /usr/local/lib/ruby/gems/3.1.0/gems/aws-sdk-core-3.131.1/lib/seahorse/client/base.rb:102:in `new'
               from /home/precision/dev/fsw2/toolchain/openc3/openc3/lib/openc3/utilities/aws_bucket.rb:29:in `initialize'
               from /home/precision/dev/fsw2/toolchain/openc3/openc3/lib/openc3/utilities/bucket.rb:33:in `new'
               from /home/precision/dev/fsw2/toolchain/openc3/openc3/lib/openc3/utilities/bucket.rb:33:in `getClient'

Regards,

David

Ce message et toutes les pièces jointes sont établis à l'intention exclusive de ses destinataires et sont confidentiels. Si vous recevez ce message par erreur, merci de le détruire et d'en avertir immédiatement l'expéditeur.

The information contained in this e-mail and any attachments are intended solely for the addressees. If you have received this email in error, please contact the sender and delete the email from your system.

Diese Nachricht und allfällige angehängte Dokumente sind vertraulich und nur für den/die Adressaten bestimmt. Sollten Sie nicht der beabsichtigte Adressat sein, bitten wir den Absender zu verständigen und die Information zu vernichten.

ryanmelt, 2023-02-21:
Thanks David.   Things just keep getting cut off before what I need to see.   What code is trying to use bucket?   Or does the stack trace end there?

DavidWathen, 2023-02-21:
The stack trace ends there.

ryanmelt, 2023-02-21:
What version of our code are you using?

ryanmelt, 2023-02-21:
Nevermind, I see the problem.

DavidWathen, 2023-02-21:
I have know idea what code is trying to use buckets. I assume this is AWS buckets you are talking about? Jason said it comes from the code that is talking to MINIO.

I've seen this both in 5.1.1 and 5.4.2. What is weird is we haven't used OpenC3 for a while and I know that we have had it working successfully on 5.1.1. I upgraded to the latest version to take advantage of some key performance improvements that you have done.

DavidWathen, 2023-02-21:
Do you still need the INTERFACE line details?

I can tell you that we are using a custom protocol that inherits from the LENGTH protocol. Could this be causing a difference?

ryanmelt, 2023-02-21:
Probably don't need the INTERFACE.  I'll let you know once I get further.

ryanmelt, 2023-02-21:
I've confirmed our difference is that you are using much older aws gems

DavidWathen, 2023-02-21:
Thanks so much for your help!

Can you point me to where these aws gems are located or how I can fix this?

ryanmelt, 2023-02-21:
You appear to be using a system installed ruby with gems stored at:
/usr/local/lib/ruby/gems/3.1.0/gems/

sudo gem install aws-sdk-s3
sudo gem install aws-sdk-kms
sudo gem install aws-sigv4

should get you updated.  I'll also make the change to our next release so that bridges will still work with older gems.

DavidWathen, 2023-02-21:
Thanks so much Ryan for figuring this out!!!

I have installed the gems and now my bridge comes up as expected. 

Wahoo!!!

PR #504 (2023-02-22): Bridge updates for bridgegem system
closes #497 
closes #354 
closes #303 

Example bridges:
https://github.com/OpenC3/openc3-cosmos-bridge-serial
https://github.com/OpenC3/openc3-cosmos-bridge-hidapi
https://github.com/OpenC3/openc3-cosmos-bridge-dualsense

Bridge documentation: https://openc3.com/docs/v5/bridges
closed 2023-02-22 by ryanmelt