OpenC3/cosmos #507: Any way to upload a gem with scripts WITHOUT creating a target?

GTI-Dakota, 2023-02-22:
Hello,

I have some scripts in a gem but do not want to generate a target for it as it has no CMD/TLM, Interfaces, etc. I do not wish to have microservices spin up for a target that will have no interaction. Is there a way to upload a Gem with folders containing scripts so I can open them in script runner?

Thanks, 

Dakota

MTI-twalker, 2023-02-23:
I am not sure if this is what you want. If you create a folder called "SCRIPTS" under the "config" bucket: "DEFAULT/targets/SCRIPTS". Then upload the scripts to that folder. It shows up in script runner. That is because "Open File" searches for all script files within all subfolders of the targets & targets_modified directory.

GTI-Dakota, 2023-02-23:
@MTI-twalker ,

Thanks for pointing that out! I had done that before and it worked fairly well, but it's not ideal for version control and auto-building a collection of gems. For now I'm just using a target with almost nothing in it.

finchlt, 2023-02-26:
Also was checking the most recent scripting best practices documentation and it still recommends a procedure folder at the plugin level. See [https://openc3.com/docs/v5/scripting-best-practices#organizing-your-scripts-into-separate-files](url). Not sure how this would work in in the v5 architecture but would definitely help scripting organization for procedures that span multiple targets.

jmthomas, 2023-03-01:
Due to the way our plugin system works with targets and targets_modified, scripts need to belong to a target. Other programs we have supported have put their scripts in a PROCEDURES target or something similar. We should add functionality to our plugins to detect a plugin with no cmd/tlm definition files and then not spawn the decommutation and other telemetry processing microservices. We also need to update the documentation!

closed 2023-03-01 by jmthomas

jmthomas, 2023-03-02:
docs updated in https://github.com/OpenC3/openc3.com/pull/43