OpenC3/cosmos #510: Make Suite runner less of an easter egg

MTI-twalker, 2023-02-23:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
Suite runner has a lot of good functionality when it comes to building out procedures. In the documentation, it appears it is only mentioned here: https://openc3.com/docs/v5/scripting-best-practices#script-runner
I am not sure if it is meant to be an easter egg or only known by experienced users.

**Describe the solution you'd like**
I am suggesting to add to "File->New Suite File" under the "New File" button.
[image]

Once clicked, a template gets added to a blank script runner:
```
# Save file to a target directory in order to reveal suite runner controls.
require 'openc3/script/suite.rb'

class TemplateGroup < OpenC3::Group
  def script_template
    # Do something
  end
  
  def setup
    # Do setup
  end
  
  def teardown
    # Do teardown
  end
end

class TemplateSuite < OpenC3::Suite
  def initialize
    super()
    add_group('TemplateGroup')
  end
  def setup
    # Do setup
  end  
  def teardown
    # Do teardown
  end
end
```

I did notice that the file needs to be saved to a target in order to view the suite runner GUI updates. So I am adding a comment to the top of the script.

jmthomas, 2023-02-23:
Not supposed to be an easter egg and we document it under the [Script Runner](https://openc3.com/docs/v5/script-runner#running-script-suites) tool docs but that doesn't help you build the framework. I like this idea as a way to get someone right into a functional Suite.

MTI-twalker, 2023-02-23:
I missed that part of the documentation when searching! Woops!

PR #514 (2023-02-26): Add new test suite option
closes #510
closed 2023-02-26 by ryanmelt