OpenC3/cosmos #518: inject_tlm invalidated due to not providing packet_time

MTI-twalker, 2023-02-28:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
inject_tlm is causing a NoMethodError.

This is due to the fact that the telemetry_topic is now expecting the attribute of packet_time but inject_tlm never populates that attribute so it is nil at the time it attempts to write to the telemetry topic line 29

**To Reproduce**
Use the inject_tlm method

**Expected behavior**
I am expecting tlm_api.rb around line 152 to define the packet_time. This packet_time should include any potential derived conversion be applied to it if it is declared in the definitions.

**Screenshots**
/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.0/lib/openc3/topics/telemetry_topic.rb:29:in `write_packet': undefined method `to_nsec_from_epoch' for nil:NilClass (NoMethodError)

        :time => packet.packet_time.to_nsec_from_epoch,
                                   ^^^^^^^^^^^^^^^^^^^

**Environment (please complete the following information):**

- OS: Windows 10
- OpenC3 COSMOS Version 5.5.0
- Browser
  -- [] Internet Explorer (IE)
  -- [X] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

jmthomas, 2023-03-01:
@MTI-twalker I can't reproduce. I tried
```
inject_tlm("EXAMPLE", "STATUS", { "VALUE" => 10 })
```
And it works fine. Can you give me an example that reproduces with our Demo?

MTI-twalker, 2023-03-01:
@jmthomas So I thought it was just the inject_tlm but it is actually due to the read_conversion that I apply to the packet. So add into recreation that you have a custom read_conversion. I was able to dig a bit more and grabbed the log from the cmd_tlm_api container:
```
2023-03-01 10:19:26 {"time":1677683966281261600,"@timestamp":"2023-03-01T15:19:26.281+00:00","severity":"ERROR","microservice_name":"CMD__TLM__API","container_name":"be0fe0ecd216","log":"PACKET_TIME read_conversion of {\"class\"=\u003e\"OpenC3::CustomTimeConversion\", \"params\"=\u003e[\"TIMESEC\", \"TIMEUS\"]} could not be instantiated due to uninitialized constant OpenC3::CustomTimeConversion"}
2023-03-01 10:19:26 I, [2023-03-01T15:19:26.296893 #892]  INFO -- : [b8558156-6ad1-4600-9152-9a3da8a862d5] Started GET "/openc3-api/plugins?scope=DEFAULT" for 192.168.48.1 at 2023-03-01 15:19:26 +0000
2023-03-01 10:19:26 I, [2023-03-01T15:19:26.300008 #905]  INFO -- : [1c461d8d-5377-4847-8ec8-db03d9b2b1ab] Completed 200 OK in 20ms (Views: 16.3ms | Allocations: 989)
2023-03-01 10:19:26 I, [2023-03-01T15:19:26.305008 #892]  INFO -- : [b8558156-6ad1-4600-9152-9a3da8a862d5] Processing by PluginsController#index as JSON
2023-03-01 10:19:26 I, [2023-03-01T15:19:26.310835 #892]  INFO -- : [b8558156-6ad1-4600-9152-9a3da8a862d5]   Parameters: {"scope"=>"DEFAULT"}
2023-03-01 10:19:26 {"time":1677683966299874700,"@timestamp":"2023-03-01T15:19:26.299+00:00","severity":"ERROR","microservice_name":"CMD__TLM__API","container_name":"be0fe0ecd216","log":"NoMethodError : undefined method `to_nsec_from_epoch' for nil:NilClass\n\n        :time =\u003e packet.packet_time.to_nsec_from_epoch,\n                                   ^^^^^^^^^^^^^^^^^^^\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/topics/telemetry_topic.rb:29:in `write_packet'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/api/tlm_api.rb:153:in `inject_tlm'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/io/json_drb.rb:267:in `public_send'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/io/json_drb.rb:267:in `process_request'\n/src/app/controllers/api_controller.rb:87:in `handle_post'\n/src/app/controllers/api_controller.rb:41:in `block in api'\n/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'\n/src/app/controllers/api_controller.rb:28:in `api'"}
```

I was able to get it to work if I used the packet_time conversion that is in the source code like unix_time_conversion.rb because it was able to load that class by default. When I introduce a custom_read_conversion script the cmd_tlm_api does not have that loaded into the thread maybe?

So to recreate, add a custom time conversion script, I copied the unix_time_conversion and renamed the file/class. Then add that as a READ_CONVERSION to your packet_time item:

custom_time_conversion.rb (unix_time_conversion.rb copied to target lib directory)
```
  class CustomTimeConversion < Conversion
```

tlm.txt, add this PACKET_TIME to a packet with TIMESEC and TIMEUS or some timestamp.
```
	ITEM PACKET_TIME 	0 0 DERIVED "Ruby time based on CCSDS_SECONDS and CCSDS_SUBSECS"	
	  READ_CONVERSION custom_time_conversion.rb TIMESEC TIMEUS  
```

Attempt to inject_tlm.

jmthomas, 2023-03-04:
I can reproduce this. The PACKET_TIME conversion is special case because we have code in the internal Packet class called packet_time which looks like this:
```
    def packet_time
      item = @items['PACKET_TIME'.freeze]
      if item
        return read_item(item, :CONVERTED, @buffer)
      else
        return @received_time
      end
    end
```
I'm not sure if there is an easy work around for this. @ryanmelt?

jmthomas, 2023-04-12:
Also it appears that inject_tlm doesn't affect the RECEIVED_COUNT of live telemetry. This is due to inject_tlm affecting the TelemetryTopic but not the interface_microservice code which handles the raw packet processing.

PR #614 (2023-04-16): inject_tlm performed by microservices
also repair interface_cmd, protocol_cmd, and router commanding
closes #518
closed 2023-04-16 by ryanmelt