OpenC3/cosmos #521: Recommended Ruby Dependency Location

mattgermano, 2023-02-28:
General Question
-------------------

Where is the recommended location to add/install a custom Ruby runtime dependency? For example, `'google-protobuf', '~> 3.22'`. 

Previously, I was adding it directly to the `openc3.gemspec` file [here](https://github.com/OpenC3/cosmos/blob/main/openc3/openc3.gemspec#L111). This requires building custom versions for the majority of the Docker images since the other images depend on `openc3-base`. 

It likely makes more sense now to add the runtime dependency to the plugin in `openc3-cosmos-demo.gemspec`, but that won't handle installing the gem.

It may work to only install the gem in the `openc3-operator/Dockerfile`. This minimizes the number of custom Docker images that need to be built.

Any recommendations?

ryanmelt, 2023-02-28:
If you are on an internet connected computer, adding dependencies to the plugin gemspec will install them.  Otherwise you can manually install other gems in the Admin -> Gems tab.

mattgermano, 2023-02-28:
Awesome. Thank you. That was easier than I thought.

closed 2023-02-28 by mattgermano