OpenC3/cosmos #522: Load Rather Than Require a Plugin's Files

jeffstjean, 2023-02-28:
**Is your feature request related to a problem? Please describe.**
The first time you install a plugin, protocol files will be correctly loaded (using the below snippet). However, since `require_class()` uses `require_file` which uses `require`, the protocol classes will not be reloaded if a plugin is installed, modified, and reinstalled. I believe this is the case with any included Protocols, Conversions, etc.

```
@protocols.each do |protocol|
  klass = OpenC3.require_class(protocol[1])
  interface_or_router.add_protocol(klass, protocol[2..-1], protocol[0].upcase.intern)
end
```

**Describe the solution you'd like**
An easy change to this would be adding an option to `require_class`: `load`.
```
self.require_class(class_name_or_class_filename, log_error = true, load: false)
  {...}
  self.require_file(class_filename, log_error, load: load)
end
```
Which can be passed to `require_file`:
```
def self.require_file(filename, log_error = true, load: false)
  if(load)
    load filename
  else
    require filename
  end
rescue Exception => err
  msg = "Unable to require #{filename} due to #{err.message}. "\
        "Ensure #{filename} is in the OpenC3 lib directory."
  Logger.error msg if log_error
  raise $!, msg, $!.backtrace
end
```
This isn't a perfect solution as now the naming of `require_class` and `require_file` aren't technically correct but it has the least impact on the existing code base. To mitigate, the `load` flag in `require_class` could either call `require_file` or a new function `load_file` which can have a more correct error message ("unable to load" rather than "unable to require").

If this is something that would be beneficial to the codebase, let me know and I can fork and submit a PR.

jmthomas, 2023-03-01:
When we upgrade plugins we uninstall and re-install them. This should kill the process and completely restart it. Thus it shouldn't matter whether we do a require or load. Do you have a specific reproducible issue that can be demonstrated with the demo or some other plugin that exposes an issue?

jeffstjean, 2023-03-02:
Ah my bad I tried reproducing with the demo plugin and couldn't. I think I just have an atypical environment set up or I was doing something else weird. Thanks!

closed 2023-03-02 by jeffstjean