OpenC3/cosmos #523: Dynamically construct content security policy

wiatrp, 2023-03-01:
**Is your feature request related to a problem? Please describe.**
`Content-Security-Policy` header in `openc3-tool-base/src/index.ejs` needs to be updated if I want to host COSMOS on a server with a custom hostname like "mymcs.com". 

**Describe the solution you'd like**
A way to dynamically construct the content security policy, maybe with env variables. I would like to add a specific hostname that content security policy would look for which would change for different environments (qa/prod/etc).

wiatrp, 2023-03-01:
Although now that I'm thinking about it more, I would think the `self` keyword would take care of that. So maybe the CSP errors I'm running into has to do with how I'm hosting/routing requests and not with the CSP itself?

ryanmelt, 2023-03-01:
The default CSP requires https for anything except localhost.  If you want to just use http, you need to change your compose.yaml to point to the -allow-http config file for traefik.  (Its a commented out line).

closed 2023-03-01 by ryanmelt