OpenC3/cosmos #531: Cable Streaming Capability

bradenvirdellgenevatech, 2023-03-06:
Hello,

It has come to my attention that one of my previously working custom widgets is no longer working. From my digging it seems like the cable subscription capability has been updated. I have tried a few different things, specifically updating the subscription key a few different ways based on what I am seeing in the Graph.vue file. None of these have succeeded so far I was wondering if you could take a look at what I currently have to see if there are any issues? 

Originally I was subscribed to an entire packet with the following:
[image]
And the subscription perform looks like this:
[image]

My updated subscription key looks like this, in this one I am trying to just subscribe to a single tlm:
[image]
And the subscription perform looks like this:
[image]

Any help would be greatly appreciated. Thank you.

jmthomas, 2023-03-06:
The `subscriptionKey` function at the bottom of Graph.vue shows the details of the key:
```
    subscriptionKey: function (item) {
      let key = `${item.reduced}__TLM__${item.targetName}__${item.packetName}__${item.itemName}__${item.valueType}`
      if (
        item.reduced === 'REDUCED_MINUTE' ||
        item.reduced === 'REDUCED_HOUR' ||
        item.reduced === 'REDUCED_DAY'
      ) {
        key += `__${item.reducedType}`
      }
      return key
    },
```
Note it's being called using map which maps each item in `this.items` to `item` in the function. Your code appears to use `this.parameters` every time which is probably the issue. The overall structure of the key looks correct.

I'm not sure what you're trying to accomplish but note that all the Graph widgets (https://openc3.com/docs/v5/telemetry-screens#linegraph) now support a HISTORY setting which displays graph history.

bradenvirdellgenevatech, 2023-03-06:
Currently I am just trying to stream a string TLM to a widget and when the TLM updates it the widget will update along with it, while holding pervious data to be viewed. 
The parameters are simply the Target, packet, item. I will try and add in the extra bits you have and see if that works. I will post again successful or not.

bradenvirdellgenevatech, 2023-03-06:
I am now able to get the wanted TLM to the widget, thank you for the pointers.

closed 2023-03-06 by bradenvirdellgenevatech