OpenC3/cosmos #532: Microservice documentation

tsjordan-eng, 2023-03-07:
I'm working with a SCPI device and want to get its status periodically which would require sending a command. I figured this should be done with a microservice but other than the directory structure and config there is not much in the documentation about actually writing one. 

Am I on the right track for periodic commands?

Is there more documentation? I looked at some in the example plugin, but I don't really know what methods I need.  I tried a simple command like `ls` and that worked, but it couldn't find any file I placed in my microservice subdirectory.

tsjordan-eng, 2023-03-07:
Some more references 
- https://github.com/OpenC3/openc3.com/issues/20
- https://github.com/OpenC3/cosmos/issues/280

I see it's already been asked, but I'm still wondering about the best way to send a command periodically.

GTI-Dakota, 2023-03-07:
Here is a quick example:

```
require 'openc3'
require 'openc3/api/api'
require 'openc3/utilities/logger'
require 'openc3/microservices/microservice'

include OpenC3::Api

$openc3_scope = <my_scope>
@logger = OpenC3::Logger.new
@logger.scope = $openc3_scope
@logger.microservice_name = "my_scpi_microservice"

loop do
    # Poll SCPI every n second
    begin
        cmd(<my_command>)
    rescue Exception => x
        @logger.error("Could not send command! #{x.message}")
    end
    sleep n
end
```

jmthomas, 2023-03-07:
Thanks for the example @GTI-Dakota! I'll also note that we added a new microservice generator in v5.5.0 that should get you started faster with a template. You use it like so:
```
openc3.sh cli generate microservice scpi
```

tsjordan-eng, 2023-03-09:
Thanks! I also added a little if statement to check if the interface is connected by parsing the `api.get_all_interface_info` so that it wouldn't try to send if it wasn't connected.

I'm not sure how all the logging stuff works (`@logger.debug()` doesn't show up but the others do) and the global scope variable (I set it to `'DEFAULT'` for now) so some docs would be good eventually.

closed 2023-03-09 by jmthomas

tsjordan-eng, 2023-03-14:
For anyone curious once I changed the logger instantiation to `@logger = OpenC3::Logger.new Logger::DEBUG` the debug messages showed up.