OpenC3/cosmos #534: Where does LOG_RAW log interface data to?

scandey, 2023-03-08:
The Plugins documentation mentions that LOG_RAW provides an interface log without any COSMOS headers. [https://openc3.com/docs/v5/plugins#log_raw](https://openc3.com/docs/v5/plugins#log_raw)

This is probably a silly question, but where do those logs end up getting stored? I can't seem to find them in the Bucket Explorer / Logs / DEFAULT area that the target logs are stored.

scandey, 2023-03-10:
> Currently it is just saved into the operator container.
> See #519

In the meantime, could you point me to where the raw interface logs are in the operator container? I had a look and couldn't find them easily.

ryanmelt, 2023-03-16:
Looking at the current code, I think they will show up at /openc3/lib/openc3/microservices.   You will have to start raw logging from a script first though.
This will be properly implemented in version 5.5.3