OpenC3/cosmos #538: openc3-cosmos-cmd-tlm-api failing to connect to openc3-redis under podman

sambarusty, 2023-03-10:
I am attempting to run the release version of the cosmos containers on centos 8.7.  I cloned the cosmos project and updated the compose.yaml file for the users as specified in the documentation. This is the only change I made to the project.  I followed the podman documentation for installing docker-compose, host config updates, and podman.socket changes. Execute ./openc3.sh run and could not get to the web page.  All containers seem to be up and running.  It looks like the openc3-cosmos-cmd-tlm-api failing to connect to openc3-redis both redis and redis-ephemral are ready waiting to accept connections.  I executed a ping from script_runner to the cmd_tlm container and it worked so the container network seems to be set up ok.  However the openc3-cosmos-cmd-tlm-api keeps rebooting and reporting the following errors:

=> Booting Puma
=> Rails 7.0.4.2 application starting in production
=> Run `bin/rails server --help` for more startup options
Exiting
/usr/lib/ruby/3.1.0/socket.rb:227:in `getaddrinfo': getaddrinfo: Try again (Redis::CannotConnectError)
        from /usr/lib/ruby/3.1.0/socket.rb:227:in `foreach'
        from /usr/lib/ruby/3.1.0/socket.rb:632:in `tcp'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client/ruby_connection.rb:49:in `initialize'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client.rb:665:in `new'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client.rb:665:in `block in connect'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client/middlewares.rb:12:in `connect'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client.rb:664:in `connect'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client.rb:658:in `raw_connection'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client.rb:625:in `ensure_connected'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-client-0.12.2/lib/redis_client.rb:224:in `call_v'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-5.0.6/lib/redis/client.rb:73:in `call_v'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-5.0.6/lib/redis.rb:167:in `block in send_command'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-5.0.6/lib/redis.rb:166:in `synchronize'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-5.0.6/lib/redis.rb:166:in `send_command'
        from /usr/lib/ruby/gems/3.1.0/gems/redis-5.0.6/lib/redis/commands/hashes.rb:26:in `hset'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/utilities/store_autoload.rb:74:in `public_send'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/utilities/store_autoload.rb:74:in `block in method_missing'
        from /usr/lib/ruby/gems/3.1.0/gems/connection_pool-2.3.0/lib/connection_pool.rb:65:in `block (2 levels) in with'
        from /usr/lib/ruby/gems/3.1.0/gems/connection_pool-2.3.0/lib/connection_pool.rb:64:in `handle_interrupt'
        from /usr/lib/ruby/gems/3.1.0/gems/connection_pool-2.3.0/lib/connection_pool.rb:64:in `block in with'
        from /usr/lib/ruby/gems/3.1.0/gems/connection_pool-2.3.0/lib/connection_pool.rb:61:in `handle_interrupt'
        from /usr/lib/ruby/gems/3.1.0/gems/connection_pool-2.3.0/lib/connection_pool.rb:61:in `with'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/utilities/store_autoload.rb:74:in `method_missing'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/utilities/store_autoload.rb:69:in `public_send'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/utilities/store_autoload.rb:69:in `method_missing'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/models/model.rb:149:in `create'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/models/model.rb:81:in `set'
        from /src/config/initializers/cts.rb:56:in `<main>'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/top_level.rb:577:in `load'
        from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.1/lib/openc3/top_level.rb:577:in `load'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/engine.rb:667:in `block in load_config_initializer'
        from /usr/lib/ruby/gems/3.1.0/gems/activesupport-7.0.4.2/lib/active_support/notifications.rb:208:in `instrument'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/engine.rb:666:in `load_config_initializer'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/engine.rb:620:in `block (2 levels) in <class:Engine>'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/engine.rb:619:in `each'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/engine.rb:619:in `block in <class:Engine>'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/initializable.rb:32:in `instance_exec'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/initializable.rb:32:in `run'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/initializable.rb:61:in `block in run_initializers'
        from /usr/lib/ruby/3.1.0/tsort.rb:228:in `block in tsort_each'
        from /usr/lib/ruby/3.1.0/tsort.rb:350:in `block (2 levels) in each_strongly_connected_component'
        from /usr/lib/ruby/3.1.0/tsort.rb:422:in `block (2 levels) in each_strongly_connected_component_from'
        from /usr/lib/ruby/3.1.0/tsort.rb:431:in `each_strongly_connected_component_from'
        from /usr/lib/ruby/3.1.0/tsort.rb:421:in `block in each_strongly_connected_component_from'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/initializable.rb:50:in `each'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/initializable.rb:50:in `tsort_each_child'
        from /usr/lib/ruby/3.1.0/tsort.rb:415:in `call'
        from /usr/lib/ruby/3.1.0/tsort.rb:415:in `each_strongly_connected_component_from'
        from /usr/lib/ruby/3.1.0/tsort.rb:349:in `block in each_strongly_connected_component'
        from /usr/lib/ruby/3.1.0/tsort.rb:347:in `each'
        from /usr/lib/ruby/3.1.0/tsort.rb:347:in `call'
        from /usr/lib/ruby/3.1.0/tsort.rb:347:in `each_strongly_connected_component'
        from /usr/lib/ruby/3.1.0/tsort.rb:226:in `tsort_each'
        from /usr/lib/ruby/3.1.0/tsort.rb:205:in `tsort_each'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/initializable.rb:60:in `run_initializers'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/application.rb:372:in `initialize!'
        from /src/config/environment.rb:5:in `<main>'
        from config.ru:3:in `require_relative'
        from config.ru:3:in `block in <main>'
        from /usr/lib/ruby/gems/3.1.0/gems/rack-2.2.6.2/lib/rack/builder.rb:116:in `eval'
        from /usr/lib/ruby/gems/3.1.0/gems/rack-2.2.6.2/lib/rack/builder.rb:116:in `new_from_string'
        from /usr/lib/ruby/gems/3.1.0/gems/rack-2.2.6.2/lib/rack/builder.rb:105:in `load_file'
        from /usr/lib/ruby/gems/3.1.0/gems/rack-2.2.6.2/lib/rack/builder.rb:66:in `parse_file'
[... 176 lines cut ...]
        from /usr/lib/ruby/gems/3.1.0/gems/thor-1.2.1/lib/thor/command.rb:27:in `run'
        from /usr/lib/ruby/gems/3.1.0/gems/thor-1.2.1/lib/thor/invocation.rb:127:in `invoke_command'
        from /usr/lib/ruby/gems/3.1.0/gems/thor-1.2.1/lib/thor.rb:392:in `dispatch'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/command/base.rb:87:in `perform'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/command.rb:48:in `invoke'
        from /usr/lib/ruby/gems/3.1.0/gems/railties-7.0.4.2/lib/rails/commands.rb:18:in `<main>'
        from /usr/lib/ruby/gems/3.1.0/gems/bootsnap-1.16.0/lib/bootsnap/load_path_cache/core_ext/kernel_require.rb:32:in `require'
        from /usr/lib/ruby/gems/3.1.0/gems/bootsnap-1.16.0/lib/bootsnap/load_path_cache/core_ext/kernel_require.rb:32:in `require'
        from bin/rails:4:in `<main>'
=> Booting Puma
=> Rails 7.0.4.2 application starting in production
=> Run `bin/rails server --help` for more startup options

**Expected behavior**
A clear and concise description of what you expected to happen.
Expected it to just come up and connect to the cmd_tlm web page like it did on windows.

**Environment (please complete the following information):**

- OS:Centos 8.7 Ootpa
- C3 COSMOS Version 5.5.1
- Browser
  -- [] MS Edge
  -- [] Google Chrome
  -- [] Mozilla Firefox

ryanmelt, 2023-03-15:
Podman 4 changed how you have to configure podman to get DNS to work.   Podman directions have been updated:
https://openc3.com/docs/v5/podman

closed 2023-03-15 by ryanmelt

sambarusty, 2023-03-16:
Thanks for the help, that was it.

Closed #538 as completed.