OpenC3/cosmos #539: Use dynamic Target definitions w/out constantly rebuilding plugins

ShaurCapella, 2023-03-11:
Howdy - in a world where we use cosmos to command multiple different assets with changing packet definitions,  in cosmos 3/4 we were able to just update the config folders target files and reload cosmos and we were good to go, cosmos read all of this in at run time. In OpenC3 whats the recommended best practice to do so, rebuilding a plugin each time feels to be quite overdone for something like adding a field to a packet def.

Thanks in advance!

jmthomas, 2023-03-11:
COSMOS 5 is container based and is changed based on installing plugins. In order to spin up the correct data decommutation, data reduction, logging, etc we need to rely on this plugin installation method. So yes this does add a little more friction during the initial plugin development process but once you have things defined everything should work as before. I'd recommend you develop your multiple plugin definitions based on your assets and install them and activate them as needed.

jmthomas, 2023-03-11:
If you truly have dynamically changing packet definitions we have another issue to discuss that: https://github.com/OpenC3/cosmos/issues/419. We're open to potential solutions.

closed 2023-03-11 by jmthomas

ShaurCapella, 2023-03-11:
Thanks for the quick response and noted! Yup we do have multiple changing packet definitions - testing the way we fly is nice but as a result also means on the R&D side we might be changing packet defs a lot more often than you'd hope! I'll comment on 419 for a thought I had. Appreciate the insight about COSMOS 5!