OpenC3/cosmos #547: How to use get_target_file() in an interface file?

GTI-Dakota, 2023-03-15:
Hello,

I am trying to use the get_target_file() script call in order to load a config file into my custom interface file so that I dont have to pass a million arguments via the plugin.txt. Here is my basic require block at the top of the interface file:

> require 'yaml'
> require 'open3'
> require 'openc3/script'
> require 'openc3/api/api'
> require 'openc3/interfaces/interface'
> 
> include OpenC3::Api

However, when the interface starts up, I get the following error:
`/usr/lib/ruby/gems/3.1.0/gems/openc3-5.5.0/lib/openc3/script/storage.rb:181:in '_get_presigned_request': Failed to get presigned URL for /openc3-api/storage/download/[MY_SCOPE]/targets/[MY_TARGET]/lib/config.yaml (RuntimeError)`

Is there something I can add to the top of my file so that I can utilize the get_target_file()?

Thanks,

Dakota

GTI-Dakota, 2023-03-15:
I forgot to mention: as a workaround I am using the openc3/utilities/aws_bucket API to get files :+1:. Not sure if this is recommended, which is why I am asking about the other method.

jmthomas, 2023-03-15:
Try `require 'openc3/script/storage'`

ryanmelt, 2023-03-17:
Using Bucket directly from Interfaces is the correct path, they have the right permissions for that.
I'll look into why get_target_file wasn't working too though.

ryanmelt, 2023-03-17:
Note that you should use Bucket not AwsBucket directly so that the code will work on all platforms.

ryanmelt, 2023-03-17:
Cannot recreate.  I replaced the example_interface.rb in the demo with this code:

```
require 'yaml'
require 'open3'
require 'openc3/script'
require 'openc3/api/api'
require 'openc3/interfaces/tcpip_client_interface'

include OpenC3::Api

class ExampleInterface < OpenC3::TcpipClientInterface
  def initialize (hostname, port, write_timeout = 10.0, read_timeout = nil)
    super(hostname, port, port, write_timeout, read_timeout, 'LENGTH', 0, 32, 4, 1, 'BIG_ENDIAN', 4, nil, nil, true)
  end

  def connect
    super
    file = get_target_file("INST/procedures/collect.rb")
    puts file.read
  end
end
```

Worked like a champ.

closed 2023-03-17 by ryanmelt