OpenC3/cosmos #559: Upgrade from 5.4.2 to 5.5.2 Caused Telemetry Grapher and Script Runner to stop working

tsjordan-eng, 2023-03-24:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Upgrade from 5.4.2 to 5.5.2 Caused Telemetry Grapher and Script Runner to stop working.

I followed this update procedure: https://openc3.com/docs/v5/upgrading#cosmos-upgrades

**Screenshots**
Telemetry Grapher has warning:
```json
[
    {
        "type": "disconnected",
        "message": "OpenC3 backend connection disconnected",
        "time": 1679695668361
    }
]
```

Script runner just hangs:
[image]

Rolling the containers and repo back fixes it.

**Environment (please complete the following information):**

- OS: RHEL 7 Docker
- OpenC3 COSMOS Version 5.5.2
- Browser
  -- [x] Edge
  -- [] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

ryanmelt, 2023-03-27:
Unable to recreate between 5.4.2 and 5.5.2 using the current cosmos-project.
I could see there being issues going from 5.3.0 to 5.4.2 because some permission changed, but don't see any obvious problems between 5.4.2 and 5.5.2.   Please let me know if you can recreate and provide any logs.
Thanks.

ryanmelt, 2023-03-31:
Closing.  Please reopen if you have this problem again.

closed 2023-03-31 by ryanmelt